chipsalliance/firrtl #941: Custom verilog emission

grebe, 2018-11-16:
**Type of issue**:  other enhancement

**What is the use case for changing the behavior?**

@azidar mentioned that you were looking at adding a special node type to firrtl, @albertchen-sifive.

I'm interested in being able to support a use-case where a chisel annotation is used to emit custom verilog. For example, I'd like to be able to emit \#ifdef guards for simulation around `Analog` types so we can use Verilog `real`.

Is this something close to what you're working on? If so, I'd like to be involved.

**Impact**: unknown

**Development Phase**: proposal

edwardcwang, 2018-11-16:
One thing to note is that we've been using `Analog` types for `inout` wires so hopefully using Verilog `real` will be an optional change/transform, not mandatory.

grebe, 2018-11-16:
I think the current analog behavior is probably better called inout (at least from chisel), but I see this as being optional in the sense that the current behavior is probably what should continue to happen by default, but there should be a mechanism for an annotation/custom node type coming out of chisel to emit custom verilog (simulation ifdefs being valuable to me). Perhaps chisel should spit this annotation out whenever you use analog and have something called InOut that doesn't use the annotation.

edwardcwang, 2018-11-16:
Sounds good. The current `Analog` seems almost like a blackbox type e.g. maybe even something like `Net`.

albertchen-sifive, 2018-11-17:
yep, I would like to add some custom node type that can be extended to emit things like #ifdef and such. I haven't gotten around to implementing anything yet though. @azidar you mentioned that there was some cleanup on some of the firrtl passes that should be done before starting?

seldridge, 2018-11-19:
I think there are two things here. You'd like to be able to add transforms that run only for a specific emitter. You'd also like to be able to override how specific nodes are emitted and to add custom nodes where you define their emission behavior.