chipsalliance/firrtl #943: Direction versus male/female

schoeberl, 2018-11-17:
**Type of issue**: other enhancement

On Bundles Chisel uses Input and Output for directions. Why introducing the terms male and female in FIRRTL? First, this introduces new terms without a real need (and needs some thinking to map). Second, it sounds a little bit sexism to me. I know we use those terms for physical connectors, but not to indicate a direction. Which is BTW another argument to not use them for a direction indication.

Anything around in/out/inout or input/output/bidirectional sounds more logical.

mwachs5, 2018-11-17:
is it always input/output? Why not just say "LHS/RHS" since that's kinda where the errors always show up?

azidar, 2018-11-17:
Male and female were used instead to refer to whether an expression was a sink or source. Its different than input/output, because a module's input is a source, but an instance's input is a sink.

Re sexism, I actually totally agree with that, but at the time I had a really hard time thinking of a better name. Since that time, I've realized that 'source' and 'sink' are much more appropriate terms and are equally as descriptive. I'd be more than happy to approve a PR that changed all references (male -> source, female -> sink).

azidar, 2018-11-17:
Oh and its not always "LHS/RHS", in that you can have a source expression on the LHS of an expression. In `x[y] := z`, y is a source.

azidar, 2018-11-17:
Although, one example of why 'source' and 'sink' aren't quite the same as 'male' and 'female' is that for FIRRTL bulk connects, the connections can actually travel left-to-right, but the left is always female and the right is always male:
```
wire foo: {flip z}
wire bar: {flip z}
foo <= bar
```
In the connection operator, `bar` is a source and `foo` is a sink, but `bar.z` is a sink, and `foo.z` is a source.

Re @mwachs5 another example where its not quite the same as "LHS/RHS" is the following:

```
wire foo: {flip z}
wire bar: {flip z}
foo.z <= bar.z
```

`foo` is male, but `foo.z` is female.

mwachs5, 2018-11-17:
sorry what is the `<=` operator? Do you mean `:=`?

azidar, 2018-11-17:
Oh in FIRRTL its `<=` not `:=` for silly reasons.

mwachs5, 2018-11-17:
ah ok. Anyway I agree with @azidar RE sexism stuff, it's not terribly offensive but source and sink are somewhat easier to understand. Not input/output.

azidar, 2018-11-17:
:P That may help. So in FIRRTL, wires can have bundles with multi-directioned fields. When connected, the direction that the fields are connected depends on whether the internal field is 'flipped' or not, and which side of the connection the reference is. The left expression starts as female, and the right expression starts as male. When you descend into flipped fields, the gender is flipped.

In the following example:
```
wire foo: {y, flip z}
wire bar: {y, flip z}
foo <= bar
```
is equivalent to:
```
wire foo: {y, flip z}
wire bar: {y, flip z}
foo.y <= bar.y
bar.z <= foo.z
```

azidar, 2018-11-17:
> What was the original direction of z?

z has no original direction - its direction (or gender) is determined by how its referenced in the connect.

mwachs5, 2018-11-17:
So In your example, let’s say I left of the “flip” for foo.

What error message would I get?

PR #1124 (2019-09-16): Gender to Flow
In light of the discussion around #943, this implements a structural rename of Working IR's `Gender` to `Flow`. I'm attempting to maintain backwards compatibility by deprecating anything gender-related that's public and providing implicit conversions where appropriate.

In summary, the following are the new names, which are assumedly more descriptive on what is going on. Nonetheless, I'm happy to bikeshed:
  - `gender        -> flow`
  - `MALE          -> SourceFlow`
  - `FEMALE        -> SinkFlow`
  - `BIGENDER      -> DuplexFlow`
  - `UNKNOWNGENDER -> UnknownFlow`

Fixes #943.
closed 2019-09-16 by seldridge