chipsalliance/firrtl #951: CircuitState constructor type mismatch

colin4124, 2018-11-27:
**Type of issue**: other enhancement

**Impact**: no functional change

**Development Phase**: proposal

`CircuitState` is defined in [Compiler.scala](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/Compiler.scala#L35)

```scala
case class CircuitState(
    circuit: Circuit,
    form: CircuitForm,
    annotations: AnnotationSeq,
    renames: Option[RenameMap]) {
...
}
```

```scala
object CircuitState {
  def apply(circuit: Circuit, form: CircuitForm): CircuitState = apply(circuit, form, Seq())
  def apply(circuit: Circuit, form: CircuitForm, annotations: AnnotationSeq): CircuitState =
    new CircuitState(circuit, form, annotations, None)
}
```

1. `def apply(circuit: Circuit, form: CircuitForm): CircuitState = apply(circuit, form, Seq())` 

The type of `annotations`  is `AnnotationSeq` , but here the value is `Seq()` , the error message is below:

[image]

2. In `CircuitState(circuit, ChirrtlForm, annos)` , `annos` 's type is `Seq[firrtl.annotations.Annotation]` don't match `AnnotationSeq`

Here the code snippet is in [Driver.scala](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/Driver.scala#L241)

```scala
  val annos = getAnnotations(optionsManager)
  
  firrtlConfig.compiler.compile(
    CircuitState(circuit, ChirrtlForm, annos),
    firrtlConfig.customTransforms
  )
```
`getAnnotations` return type is `Seq[Annotation]` which don't match `AnnotationSeq`, is defined in [Driver.scala](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/Driver.scala#L88)

```scala
def getAnnotations(
      optionsManager: ExecutionOptionsManager with HasFirrtlOptions
): Seq[Annotation] = {
  ...
}
```

Here is the error message:

[image]

I am learning the source code of FIRRTL, I copy some snippet from the `master` branch and run it. So I found these error. But when I run `sbt test` in the original FIRRTL, it passes. I can't reproduce that. Do you have some magic code to implicitly do  type conversion?

seldridge, 2018-11-27:
Your intuition about implicit conversions is correct. That specific implicit conversions for `Seq[Annotation]` to `AnnotationSeq` is here:
  - https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/package.scala#L7

colin4124, 2018-11-27:
Thanks so much!

closed 2018-11-27 by colin4124