chipsalliance/firrtl #957: 2 error occured when doing sbt test

Mysterysister, 2018-12-03:
Hi,
    I just start using Rocket chip and when I start to build the environment,after I follow the tutorial,these errors always occur,but i dont know the exact meaing of these and how to handle it

`[error] Failed tests:
[error] 	firrtlTests.transforms.CombineCatsSpec
[error] 	firrtlTests.ConstantPropagationEquivalenceSpec
`
Any help would be appreciated.

Thank You  : )

seldridge, 2018-12-03:
See: https://github.com/freechipsproject/firrtl/issues/955

The `ConstantPropagationEquivalenceSpec` requires that you have `yosys` installed for doing formal equivalence checking. If you're just doing work with Rocket Chip, you can safely ignore this.

Mysterysister, 2018-12-03:
@seldridge after i install the yosys problems still there, and my work is with the branch of Rocket-chip :Boom-template,when I'm makeing in the folder `vsim`  , some error occured:

> Building TestHarness for an ExampleBoomSystem.

Interrupt map (2 harts 2 interrupts):
  [1, 2] => dut

[error] (run-main-0) java.lang.reflect.InvocationTargetException
[error] java.lang.reflect.InvocationTargetException
[error] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities.$anonfun$elaborate$1(GeneratorUtils.scala:52)
[error] 	at chisel3.core.Module$.do_apply(Module.scala:49)
[error] 	at chisel3.Driver$.$anonfun$elaborate$1(Driver.scala:93)
[error] 	at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:298)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at chisel3.internal.Builder$.build(Builder.scala:296)
[error] 	at chisel3.Driver$.elaborate(Driver.scala:93)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities.elaborate(GeneratorUtils.scala:57)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities.elaborate$(GeneratorUtils.scala:48)
[error] 	at boom.system.Generator$.elaborate(Generator.scala:13)
[error] 	at freechips.rocketchip.util.GeneratorApp.circuit(GeneratorUtils.scala:94)
[error] 	at freechips.rocketchip.util.GeneratorApp.circuit$(GeneratorUtils.scala:94)
[error] 	at boom.system.Generator$.circuit$lzycompute(Generator.scala:13)
[error] 	at boom.system.Generator$.circuit(Generator.scala:13)
[error] 	at freechips.rocketchip.util.GeneratorApp.generateFirrtl(GeneratorUtils.scala:100)
[error] 	at freechips.rocketchip.util.GeneratorApp.generateFirrtl$(GeneratorUtils.scala:99)
[error] 	at boom.system.Generator$.generateFirrtl(Generator.scala:13)
[error] 	at boom.system.Generator$.delayedEndpoint$boom$system$Generator$1(Generator.scala:95)
[error] 	at boom.system.Generator$delayedInit$body.apply(Generator.scala:13)
[error] 	at scala.Function0.apply$mcV$sp(Function0.scala:34)
[error] 	at scala.Function0.apply$mcV$sp$(Function0.scala:34)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at scala.App.$anonfun$main$1$adapted(App.scala:76)
[error] 	at scala.collection.immutable.List.foreach(List.scala:389)
[error] 	at scala.App.main(App.scala:76)
[error] 	at scala.App.main$(App.scala:74)
[error] 	at boom.system.Generator$.main(Generator.scala:13)
[error] 	at boom.system.Generator.main(Generator.scala)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.lang.reflect.Method.invoke(Method.java:498)
[error] 	at sbt.Run.invokeMain(Run.scala:93)
[error] 	at sbt.Run.run0(Run.scala:87)
[error] 	at sbt.Run.execute$1(Run.scala:65)
[error] 	at sbt.Run.$anonfun$run$4(Run.scala:77)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] 	at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] Caused by: java.lang.NoSuchMethodError: org.json4s.JsonDSL$.seq2jvalue(Lscala/collection/Iterable;Lscala/Function1;)Lorg/json4s/JsonAST$JArray;
[error] 	at freechips.rocketchip.util.GenRegDescsAnno$.$anonfun$serialize$9(Annotations.scala:309)
[error] 	at org.json4s.JsonDSL.pair2jvalue(JsonDSL.scala:82)
[error] 	at org.json4s.JsonDSL.pair2jvalue$(JsonDSL.scala:82)
[error] 	at org.json4s.JsonDSL$.pair2jvalue(JsonDSL.scala:64)
[error] 	at freechips.rocketchip.util.GenRegDescsAnno$.serialize(Annotations.scala:309)
[error] 	at freechips.rocketchip.tilelink.TLRegisterNode.genRegDescsJson(RegisterRouter.scala:96)
[error] 	at freechips.rocketchip.tilelink.TLRegisterNode.regmap(RegisterRouter.scala:87)
[error] 	at freechips.rocketchip.devices.tilelink.TLPLIC$$anon$2.<init>(Plic.scala:287)
[error] 	at freechips.rocketchip.devices.tilelink.TLPLIC.module$lzycompute(Plic.scala:114)
[error] 	at freechips.rocketchip.devices.tilelink.TLPLIC.module(Plic.scala:114)
[error] 	at freechips.rocketchip.devices.tilelink.TLPLIC.module(Plic.scala:66)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$2(LazyModule.scala:158)
[error] 	at chisel3.core.Module$.do_apply(Module.scala:49)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:158)
[error] 	at scala.collection.immutable.List.flatMap(List.scala:335)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:156)
[... 52 lines cut ...]
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] 	at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] 	at sbt.Run$.executeTrapExit(Run.scala:124)
[error] 	at sbt.Run.run(Run.scala:77)
[... cut ...]
[error] Total time: 7 s, completed Dec 3, 2018 11:02:15 PM

> /opt/riscv/riscv-tools/riscv-gnu-toolchain/boom-template/Makefrag:29: recipe for target '/opt/riscv/riscv-tools/riscv-gnu-toolchain/boom-template/vsim/generated-src/boom.system.TestHarness.BoomConfig.fir' failed
make: *** [/opt/riscv/riscv-tools/riscv-gnu-toolchain/boom-template/vsim/generated-src/boom.system.TestHarness.BoomConfig.fir] Error 1

and how can i solve these errors

Thank you for your  reply and hoping these appendix questions will not bother you   ： ）

seldridge, 2018-12-03:
The specific error you're hitting appears to be:
```
[error] Caused by: java.lang.NoSuchMethodError: org.json4s.JsonDSL$.seq2jvalue(Lscala/collection/Iterable;Lscala/Function1;)Lorg/json4s/JsonAST$JArray;
```

I'm not sure what's going on here. This appears to be some version incompatibility across versions json4s. However, that's just me guessing based on some similarity to: https://github.com/json4s/json4s/issues/212. I was not able to replicate on my end using boom-template. A fresh clone worked, but this is using some locally published versions of Chisel3 and FIRRTL that I didn't change before testing.

I'd try a complete re-clone of boom-template and see if that does it for you.

closed 2018-12-04 by seldridge