chipsalliance/firrtl #968: Spec says cat(SInt, UInt) is valid, but compiler rejects it

aswaterman, 2018-12-14:
Currently, the Chisel concatenation operator `##` accepts any combination of UInt and SInt.  This gets translated directly to Firrtl `cat`.  The Firrtl spec claims this is valid, but the implementation issues an error when the types don't match. It does accept both UInt/UInt or SInt/SInt, and in both cases returns UInt, as the spec says it should.

@jackkoenig tells me he thinks the implementation is correct and the spec is outdated.  If that is indeed the case, we should change Chisel's definition of `##` to `asUInt` its arguments.

aswaterman, 2018-12-14:
BTW, my take is that cat should accept all combinations (and continue to return UInt unconditionally).  This doesn't add much complexity, because there's no special handling for UInt vs. SInt in this operator.

seldridge, 2018-12-18:
Tentative discussed solution: 
  - Fix the spec to disallow this
  - Update Chisel to emit casts

albert-magyar, 2020-03-12:
Closed by #1085

closed 2020-03-12 by albert-magyar