chipsalliance/firrtl #970: Refactor firrtlTests under firrtl package

seldridge, 2018-12-14:
**Type of issue**: other enhancement

The separation of `firrtlTests` from `firrtl` results in some annoyances (or lots of monkeying around) to test private behavior. E.g., consider:

```scala
package firrtl

class X private [firrtl] 
```

If I can access this constructor from `firrtl`, it seems logical that I'd like to test it's behavior in `firrtlTests`. However, by placing all the tests inside `firrtlTests`, this is private to testing. There's some hackery you can do to get at private methods, but that seems inelegant.

As far as I understand it, there's no need to place tests in a separate package. Those jars are built separately and not included in any fat jar assembly or publishing (as far as I'm aware).

I'd go one step further with this and suggest that the testing packages should match the source packages exactly in terms of names and directory structure.

**What is the use case for changing the behavior?**

Make it so that you can test private things in package `firrtl` or sub-packages.

**Impact**: no functional change

**Development Phase**: proposal

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

This problem exists for some of the other BIG6 packages (Chisel notably).

chick, 2018-12-14:
I believe this is the correct pattern and is one that the interpreter and treadle both follow

seldridge, 2018-12-14:
Yeah, this appears to only be a predominant problem in Chisel and FIRRTL. Annoyingly, I expect that fixing this will cause heinous merge conflicts with all outstanding PRs/dev work...

seldridge, 2018-12-16:
Looking around online, I'm actually not sure if what I'm proposing here is a pattern or an anti-pattern. I want to do this to enable testing of private methods. However, there seems to be a lot of disagreement on whether or not you should test private things. Or: should testing be limited to the public API only?

See: https://stackoverflow.com/questions/105007/should-i-test-private-methods-or-only-public-ones

@ducky64: Do you have any thoughts, here?

chick, 2018-12-17:
There can be large chunks of complicated code that can be completely hidden behind an API. I would not feel comfortable just believing the API is going to ensure that code works, and furthermore, if you believe in TDD at all shouldn't you be writing all your code with tests to guide you. Rant over. Place me in the test everything camp

seldridge, 2018-12-18:
I agree with both the TDD philosophy argument and the "this is ridiculous" argument. I'll update this to use @jackkoenig's `firrtl.tests` idea.

ducky64, 2018-12-18:
If I remember correctly, one of the reasons for the chisel tests being in a separate chiselTests package was so that the tests could only use the public API. Though this was probably more for documentation value and testing API completeness (that you didn't forget to expose some API in your main code). I think this was in the context of the chisel3 package object refactor too.

I think there are good arguments for testing private functions (especially since unit testing is supposed to test individual parts and provide good bug localization power). But all things equivalent (which is _not_ the assumption here), it should be preferable to test public APIs.

seldridge, 2018-12-18:
Thanks for the background info on `chiselTests`, @ducky64. I'm similarly struggling with the notion that if you want to test something, it may indicate that it's something that should be public. Additionally, it becomes less clear on what boundary you're supposed to test once you expose the internals to tests. Furthermore, changes to the private internals likely then break a bunch of tests the developer has to go fixup.

As one motivating counter example: as part of #920, I introduce a [compatibility layer](https://github.com/freechipsproject/firrtl/pull/920/files#diff-a8449cd2e78f422a34c4ccf4d2f24ce0R28) between `Driver` and `FirrtlStage`. It doesn't seem to make sense to me that this should ever be public. It has egregiously complex behavior and I never want anybody to ever find this in the API doc and start using it. However, it has to be public so that I can test it.

I can use [`PrivateMethodTester`](http://www.scalatest.org/user_guide/other_goodies#privateMethodTester) to get around this somewhat. However, my understanding of this is that it only lets you test private methods, not fully private classes/objects.

Part of the overall motivation for changing this is due to the direction of the BIG6 to use deprecation judiciously and to prevent binary incompatibility. For new code, it's a lot easier to manage this by limiting what is public from the beginning. However, that then makes testing a pain. It would seem that limiting deprecation and binary incompatibility trumps the testing methodology risks of exposing private things to testing.

Basically, our cost of going from `public` -> `private` or `public` -> `binary incompatible public` is orders of magnitude higher than going `private` -> `public`.

This is likely a developer meeting discussion...

jackkoenig, 2018-12-19:
Fair point, we can just do that as necessary

seldridge, 2018-12-20:
Closing in favor of fine-grained packaging on a case-by-case basis.

closed 2018-12-20 by seldridge