chipsalliance/firrtl #973: Internal FIRRTL Error

maloneymr, 2018-12-15:
**Type of issue**: bug report

I was trying to write a VGA controller as an exercise. I managed to trigger an internal compiler error in FIRRTL. I don't know if this is legitimately an internal error or if I didn't initialize something, but either way, some guidance would be appreciated.

```
import chisel3._
import chisel3.util.Enum

class Top extends Module {
  val io = IO(new Bundle {
    val led = Output(UInt(16.W))

    val vga_red = Output(UInt(4.W))
    val vga_green = Output(UInt(4.W))
    val vga_blue = Output(UInt(4.W))

    val vga_hs = Output(Bool())
    val vga_vs = Output(Bool())
  })

  val vga = Module(new Vga)
  val vgaClock = Module(new VgaClock);
  val counter = RegInit(UInt(32.W), 0.U)

  vga.io.vgaClock := vgaClock.io.out

  when(vgaClock.io.out) {
    counter := counter + 1.U
  }

  io.led := counter(31, 16)

  io.vga_red := vga.io.red
  io.vga_green := vga.io.green
  io.vga_blue := vga.io.blue
  io.vga_hs := vga.io.hs
  io.vga_vs := vga.io.vs
}

class Vga extends Module {
  val io = IO(new Bundle {
    val vgaClock = Input(Bool())

    val red = Output(UInt(4.W))
    val green = Output(UInt(4.W))
    val blue = Output(UInt(4.W))
    val hs = Output(Bool())
    val vs = Output(Bool())
  })

  val drawMode :: hsMode :: vsMode :: Nil = Enum(3)

  val pixelX = RegInit(UInt(10.W), 0.U)
  val pixelY = RegInit(UInt(10.W), 0.U)
  val mode = RegInit(UInt(2.W), drawMode)
  val sync = RegInit(UInt(8.W), 0.U)

  val height = 640.U
  val width = 480.U

  // in ticks
  val horizontalFrontPorch = 16.U
  val horizontalSync = 128.U
  val horizontalBackPorch = 16.U
  val horizontalBlanking = horizontalFrontPorch + horizontalSync + horizontalBackPorch
  val lineSize = horizontalBlanking

  // in lines
  val verticalFrontPorch = 10.U
  val verticalSync = 2.U
[... 297 lines cut ...]
[error] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.lang.reflect.Method.invoke(Method.java:498)
[error] 	at sbt.Run.invokeMain(Run.scala:93)
[error] 	at sbt.Run.run0(Run.scala:87)
[error] 	at sbt.Run.execute$1(Run.scala:65)
[error] 	at sbt.Run.$anonfun$run$4(Run.scala:77)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] 	at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] 	at sbt.Run$.executeTrapExit(Run.scala:124)
[error] 	at sbt.Run.run(Run.scala:77)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error] 	at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.Try$.apply(Try.scala:209)
[error] 	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 8 s, completed Dec 15, 2018 8:42:57 AM
```

maloneymr, 2018-12-15:
The offending line seems to be:

```
        io.hs := !(
          sync >= horizontalFrontPorch &&
          sync < horizontalBlanking - horizontalBackPorch
        )
```

It's not clear to me why this is causing issues, since it should be allowed under Chisel's "last assignment statement wins" rule.

Any clarity to this issue (bearing in mind I'm a beginner!) would still be appreciated!

seldridge, 2018-12-16:
Thanks for figuring out the offending line. I've stripped this down to a further MWE below:

```scala
package examples

import chisel3._
import chisel3.experimental.RawModule

import org.scalatest.{FlatSpec, Matchers}

class f973 extends FlatSpec with Matchers {

  class Foo extends RawModule {
    val x = Wire(Bool())
    when (true.B) { x := true.B }
    x := true.B
  }

  class Bar extends RawModule {
    val x = WireInit(false.B)
    when (true.B) { x := true.B }
    x := true.B
  }

  behavior of "f973 issue"

  it should "work on a Wire" in {
    chisel3.Driver.execute(Array("-X", "middle"), () => new Foo) /* This is failing */
  }

  it should "work on a WireInit" in {
    chisel3.Driver.execute(Array("-X", "middle"), () => new Bar) /* This works */
  }

}
```

The issue appears to be that `passes.ExpandWhens` will emit `WVoid` for situations where it can't resolve a default value. However, you've exposed a situation where the `when` block should be entirely removed by last connect semantics and that `WVoid` doesn't matter.

Before `ExpandWhens` the circuit is:
```
Circuit:
circuit f973Foo :
  module f973Foo :
  
    wire x : UInt<1> @[f973.scala 11:17]
    when UInt<1>("h1") : @[f973.scala 12:19]
      x <= UInt<1>("h0") @[f973.scala 12:23]
    x <= UInt<1>("h1") @[f973.scala 13:7]
```

After `ExpandWhens` the circuit is:
```
Circuit:
circuit f973Foo :
  module f973Foo :
  
    wire x : UInt<1> @[f973.scala 11:17]
    node _GEN_0 = mux(UInt<1>("h1"), UInt<1>("h0"), VOID) @[f973.scala 12:19]
    x <= UInt<1>("h1") @[f973.scala 12:23 f973.scala 13:7]
```

This then fails `ResolveKinds`.

@maloneymr: Your code, while entirely legal, should likely not be specifying a default value _last_. Last connect semantics mean that the assignment to `io.hs` in your `when` block will always be overridden by the default value you specify at the end. You likely intended to specify this before everything else. If you do that, then your code should compile. Unfortunately, this appears to be exposing a bug in FIRRTL...