chipsalliance/firrtl #977: Internal Error

schoeberl, 2018-12-19:
**Type of issue**: bug report

[info] [2.699] Done elaborating.
[error] (run-main-0) firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error] firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error] 	at firrtl.Utils$.error(Utils.scala:396)
[error] 	at firrtl.Utils$.throwInternalError(Utils.scala:178)
[error] 	at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:237)
[error] 	at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:209)
[error] 	at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:138)

Chisel version 3.1.5

OS: macOS

To reproduce:
git clone https://github.com/leros-dev/leros.git
checkout version 45d299a0ae0ea98bcd3308a4f1ff7445f8eceefb
run make

schoeberl, 2018-12-19:
I found my Chisel error/mistake that triggered it:
https://github.com/leros-dev/leros/blob/45d299a0ae0ea98bcd3308a4f1ff7445f8eceefb/src/main/scala/leros/Decode.scala#L127

So it is NOT critical for me. But maybe it is worth something (or maybe already known issue).

seldridge, 2018-12-19:
Clarifying without running your code: this is a duplicate of https://github.com/freechipsproject/firrtl/issues/852, yes? Use of a when construct to assign to a not-fully-initialized hardware component that is later overwritten?

We should probably clarify the `throwInternalError` stuff (see: #974) to indicate that the originating exception is one exception down. `throwInternalError` is used a number of places to try and indicate something that we think really bad happened. However, it's also used to wrap any unexpected exceptions that show up during compilation. Unfortunately, this means that the first block of the stack trave provides very limited information on what happened...

edwardcwang, 2018-12-19:
As Schulyler noted, it would be useful to see the stacktrace that follows.

azidar, 2019-02-28:
Should be fixed now with merging of #1039 , can you verify it fixes your issue?

schoeberl, 2019-03-07:
Yes, I can do this. What is the procedure for this? Do I need to do a local clone from FIRRTL, build it, publish it local, and reference it how to be picked up by Chisel? Or do I also need to have a local published version of Chisel?

Or does the fix live in a SNAPSHOT on the server?

Cheers,
Martin

schoeberl, 2019-03-14:
Retested with actual version and is fixed now.

closed 2019-03-14 by schoeberl