chipsalliance/firrtl #978: Repeated Testing Exhausts Metaspace

seldridge, 2018-12-19:
**Type of issue**: bug report

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

1. Open up an `sbt` shell
2. Run `test`
3. `goto` (2)

  * **What is the current behavior?**

Eventually, your `sbt` shell will hang because you ran out of metaspace. The originating exception may be pretty far up, but looks like:

```
[error] java.util.concurrent.ExecutionException: java.lang.OutOfMemoryError: Metaspace
[error] 	at java.util.concurrent.FutureTask.report(FutureTask.java:122)
[error] 	at java.util.concurrent.FutureTask.get(FutureTask.java:192)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.take(ConcurrentRestrictions.scala:213)
[error] 	at sbt.Execute.next$1(Execute.scala:110)
[error] 	at sbt.Execute.processAll(Execute.scala:113)
[error] 	at sbt.Execute.runKeep(Execute.scala:90)
[error] 	at sbt.EvaluateTask$.liftedTree1$1(EvaluateTask.scala:420)
```

This is truncated as there's really no useful information here (as far as I'm aware).

Alternatively, you sometimes get the prettier:

```
Exception in thread "Timer-3" java.lang.OutOfMemoryError: Metaspace
[info] shutting down server
java.lang.OutOfMemoryError: Metaspace
Error during sbt execution: java.lang.OutOfMemoryError: Metaspace
```

  * **What is the expected behavior?**

Repeated testing should work.

  * **Please tell us about your environment:**
    (examples)
    - version: 3655ae091249a72bd424073cfb4a382a5ab170c6
    - OS: `Linux knight 4.4.0-92-generic #115-Ubuntu SMP Thu Aug 10 09:04:33 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`
    - Scala: 2.12.7 or 2.11.12
    - `sbt`: 1.2.7 

**What is the use case for changing the behavior?**

This only really matters for developers as far as I'm aware. I like to keep an `sbt` session open and repeatedly run `test` if I'm dealing with some patch that has far reaching implications. I can avoid this with invoking `sbt test` explicitly, but then you get that `sbt` startup time.

This likely has no impact on FIRRTL users, but I can't be sure if large designs or designs that pull in lots of additional classes (custom transforms?) may be able to run out of metaspace in a single run. Anecdotally, this happens sooner for #920.

**Impact**: no functional change

**Development Phase**: request

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

This could be a number of things:
  - A bug in `sbt`
  - A bug in `scala`
  - A bug in some library we're using
  - A bug in a *specific version* of one of the above
  - A bug from the interactions of the *specific versions* of the product of the above
  - A memory leak in FIRRTL

There's some discussion on maybe the same thing here: https://github.com/sbt/sbt/issues/4166

The typical `non-solution` here is to fork tests by default so that whatever is going on here is limited to a forked JVM. *I'm hesitant to fully adopt this only because of a concern that this may be a FIRRTL memory leak.* **I highly, highly expect that it's a FIRRTL leak than anything else... if only because that's the least naive assumption.**

For some graphical information, it helps to attach `visualvm` to a running `sbt` session. That gets me the following plots:

### CPU/GC Activity
[image]

### Classes Loaded
[image]

### Heap Usage
[image]

### Metaspace
[image]

schoeberl, 2019-01-11:
> On 19 Dec, 2018, at 8:12, Schuyler Eldridge <notifications@github.com> wrote:
> 
> What is the use case for changing the behavior?
> 
> This only really matters for developers as far as I'm aware. I like to keep an sbt session open and repeatedly run test if I'm dealing with some patch that has far reaching implications. I can avoid this with invoking sbt test explicitly, but then you get that sbt startup time.
> 
> This likely has no impact on FIRRTL users, but I can't be sure if large designs or designs that pull in lots of additional classes (custom transforms?) may be able to run out of metaspace in a single run. Anecdotally, this happens sooner for #920 <https://github.com/freechipsproject/firrtl/pull/920>.
> 
> 

Do you mean “only for Chisel (the library) developer”? It may also be an issue for Chisel hardware developer (= Chisel library users). It happened to at some time as well. I assume this is an sbt thing and not a Chisel issue.

Cheers,
Martin

seldridge, 2019-01-24:
@schoeberl: yes, this should be an issue for anyone that is repeatedly running Chisel or FIRRTL in the same sbt shell without closing it. I'm not aware of most users that are doing this. However, that used to be a suggestion to speed up build times and it's valid to do.

I need to dig back into this and try to figure out what's going on...

PR #1002 (2019-02-01): Fork all sbt tasks
This avoids an apparent problem somewhere in sbt and/or scalatest
where the JVM runs out of metaspace if you repeatedly run tasks. This
is an annoyance for FIRRTL developers as you have to frequently
restart sbt. This kludges around that by forking the tests into a
separate JVM.

More information:
  - https://github.com/scalatest/scalatest/issues/1175
  - https://github.com/sbt/sbt/issues/4166

Fixes #978 (for now...)
closed 2019-02-01 by seldridge