chipsalliance/firrtl #981: Firrtl Internal Error when generating verilog

MikuChaser, 2018-12-25:
I have written some modules respectively and I can generate their verilog files. But when I connected them in a top file, the .fir file was generated successfully while verilog file was NOT.
`[error] (run-main-0) firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error] firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error]         at firrtl.Utils$.error(Utils.scala:438)
[error]         at firrtl.Utils$.throwInternalError(Utils.scala:181)
[error]         at firrtl.Driver$.$anonfun$execute$1(Driver.scala:251)
[error]         at logger.Logger$.$anonfun$makeScope$2(Logger.scala:138)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error]         at logger.Logger$.makeScope(Logger.scala:136)
[error]         at firrtl.Driver$.execute(Driver.scala:223)
[error]         at chisel3.Driver$.execute(Driver.scala:228)
[error]         at chisel3.Driver$.execute(Driver.scala:248)
[error]         at yolo.TopMain$.delayedEndpoint$yolo$TopMain$1(toptest.scala:8)
[error]         at yolo.TopMain$delayedInit$body.apply(toptest.scala:7)
[error]         at scala.Function0.apply$mcV$sp(Function0.scala:34)
[error]         at scala.Function0.apply$mcV$sp$(Function0.scala:34)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error]         at scala.App.$anonfun$main$1$adapted(App.scala:76)
[error]         at scala.collection.immutable.List.foreach(List.scala:389)
[error]         at scala.App.main(App.scala:76)
[error]         at scala.App.main$(App.scala:74)
[error]         at yolo.TopMain$.main(toptest.scala:7)
[error]         at yolo.TopMain.main(toptest.scala)
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error]         at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.lang.reflect.Method.invoke(Method.java:498)
[error]         at sbt.Run.invokeMain(Run.scala:93)
[error]         at sbt.Run.run0(Run.scala:87)
[error]         at sbt.Run.execute$1(Run.scala:65)
[error]         at sbt.Run.$anonfun$run$4(Run.scala:77)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error]         at sbt.TrapExit$App.run(TrapExit.scala:252)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] Caused by: java.util.NoSuchElementException: key not found: _T
[error]         at scala.collection.MapLike.default(MapLike.scala:232)
[error]         at scala.collection.MapLike.default$(MapLike.scala:231)
[error]         at scala.collection.AbstractMap.default(Map.scala:59)
[error]         at scala.collection.MapLike.apply(MapLike.scala:141)
[error]         at scala.collection.MapLike.apply$(MapLike.scala:140)
[error]         at scala.collection.AbstractMap.apply(Map.scala:59)
[error]         at firrtl.passes.InferTypes$.infer_types_e$1(InferTypes.scala:27)
[error]         at firrtl.passes.InferTypes$.$anonfun$run$6(InferTypes.scala:26)
[error]         at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:234)
[error]         at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:52)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at scala.collection.TraversableLike.map(TraversableLike.scala:234)
[error]         at scala.collection.TraversableLike.map$(TraversableLike.scala:227)
[error]         at scala.collection.AbstractTraversable.map(Traversable.scala:104)
[error]         at firrtl.ir.DoPrim.mapExpr(IR.scala:192)
[error]         at firrtl.Mappers$ExprMagnet$$anon$6.map(Mappers.scala:42)
[error]         at firrtl.Mappers$ExprMap$.map$extension(Mappers.scala:52)
[error]         at firrtl.passes.InferTypes$.infer_types_e$1(InferTypes.scala:26)
[error]         at firrtl.passes.InferTypes$.$anonfun$run$8(InferTypes.scala:47)
[error]         at firrtl.ir.DefNode.mapExpr(IR.scala:270)
[error]         at firrtl.Mappers$StmtMagnet$$anon$2.map(Mappers.scala:19)
[error]         at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:33)
[error]         at firrtl.passes.InferTypes$.infer_types_s$1(InferTypes.scala:47)
[error]         at firrtl.passes.InferTypes$.$anonfun$run$14(InferTypes.scala:59)
[error]         at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:234)
[error]         at scala.collection.Iterator.foreach(Iterator.scala:944)
[error]         at scala.collection.Iterator.foreach$(Iterator.scala:944)
[error]         at scala.collection.AbstractIterator.foreach(Iterator.scala:1432)
[error]         at scala.collection.IterableLike.foreach(IterableLike.scala:71)
[error]         at scala.collection.IterableLike.foreach$(IterableLike.scala:70)
[error]         at scala.collection.AbstractIterable.foreach(Iterable.scala:54)
[error]         at scala.collection.TraversableLike.map(TraversableLike.scala:234)
[error]         at scala.collection.TraversableLike.map$(TraversableLike.scala:227)
[... 64 lines cut ...]
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error]         at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.lang.reflect.Method.invoke(Method.java:498)
[error]         at sbt.Run.invokeMain(Run.scala:93)
[error]         at sbt.Run.run0(Run.scala:87)
[error]         at sbt.Run.execute$1(Run.scala:65)
[error]         at sbt.Run.$anonfun$run$4(Run.scala:77)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error]         at sbt.TrapExit$App.run(TrapExit.scala:252)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error]         at sbt.Run$.executeTrapExit(Run.scala:124)
[error]         at sbt.Run.run(Run.scala:77)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error]         at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at scala.util.Try$.apply(Try.scala:209)
[error]         at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error]         at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error]         at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] (Test / runMain) Nonzero exit code: 1
[error] Total time: 19 s, completed Dec 25, 2018 4:16:50 PM`

MikuChaser, 2018-12-25:
I have solved it. But I found another problem.
If I define a constant like "val DEPTH = 1.U << 15", there will be a error when generating verilog. So I have to change the constant to "val DEPTH = 0x8000.U(16.W)".

edwardcwang, 2018-12-25:
Can you share a snippet of the Chisel code that generated the error originally?

MikuChaser, 2018-12-25:
> Can you share a snippet of the Chisel code that generated the error originally?

Yes.
Original version.
`......
object WeightBufferParameter {
  val WRITE_DEPTH = 1.U << 15
  val WRITE_DEPTH_HALF = WRITE_DEPTH >> 1
}
......
import WeightBufferParameter._
val addr_gen = Reg(UInt(15.W))
when(reset.toBool || io.pad_fillHead) {
    addr_gen := 0.U
  } .elsewhen(io.pad_fillTail) {
    addr_gen := WRITE_DEPTH_HALF
  } .elsewhen(io.ddr_weight_en || !fillCount(14).toBool) {
    addr_gen := addr_gen + 1.U
  }
......`
If I compile this file separately, it works. But when I instantiate the module in a top file, it fails. So I have to change WRITE_DEPTH to 0x8000.U(16.W).

edwardcwang, 2018-12-25:
One side note: in Chisel, registers can be explicitly given reset values. So instead of 

```scala
val addr_gen = Reg(UInt(15.W))
when(reset.toBool || io.pad_fillHead) {
    addr_gen := 0.U
  }
```

You can just do

```scala
val addr_gen = RegInit(0.U(15.W))
```

In general in Chisel we try to avoid referencing `reset` explicitly except in `withClockAndReset` or to wire up BlackBox.

MikuChaser, 2018-12-25:
> One side note: in Chisel, registers can be explicitly given reset values. So instead of
>
> ```scala
> val addr_gen = Reg(UInt(15.W))
> when(reset.toBool || io.pad_fillHead) {
>     addr_gen := 0.U
>   }
> ``` [...]

But the error happened in
`addr_gen := WRITE_DEPTH_HALF`

edwardcwang, 2018-12-25:
Yes, I'm looking at it. Just wanted to point out an (unrelated) side note.

edwardcwang, 2018-12-25:
The following snippet of code works fine for me in Chisel 3.1.0. What version of Chisel/FIRRTL are you using?

```scala
object WeightBufferParameter {
  val WRITE_DEPTH = 1.U << 15
  val WRITE_DEPTH_HALF = WRITE_DEPTH >> 1
}

class TestMod extends Module {
  val io = IO(new Bundle {
    val pad_fillHead = Input(Bool())
    val pad_fillTail = Input(Bool())
    val ddr_weight_en = Input(Bool())
    val addr_out = Output(UInt(15.W))
  })

import WeightBufferParameter._
val addr_gen = RegInit(0.U(15.W))
when(io.pad_fillHead) {
    addr_gen := 0.U
  } .elsewhen(io.pad_fillTail) {
    addr_gen := WRITE_DEPTH_HALF
  } .elsewhen(io.ddr_weight_en /*|| !fillCount(14).toBool*/) {
    addr_gen := addr_gen + 1.U
  }

io.addr_out := addr_gen

}
```

MikuChaser, 2018-12-25:
> The following snippet of code works fine for me in Chisel 3.1.0. What version of Chisel/FIRRTL are you using?
>
> ```scala
> object WeightBufferParameter {
>   val WRITE_DEPTH = 1.U << 15
>   val [...]

Chisel 3_2.12 & Firrtl 2.12 
It can't work when I instantiate it in a top file rather than compile it solely.

edwardcwang, 2018-12-25:
`2.12` is the scala version, not the Chisel version. What is the Chisel version you are using?

Also, what do you mean by "instantiate it in a top file"? Do you mean instantiating the module in another module? Can you provide a full example so we can debug? Thanks in advance.

MikuChaser, 2018-12-25:
> `2.12` is the scala version, not the Chisel version. What is the Chisel version you are using? If you are using the chisel-template you can find it here: https://github.com/freechipsproject/chisel-template/blob/release/build.sbt#L44
> 
> Also, what do you mean by "instantiate it in a top file"? Do you mean instantiating the module in another module? Can you provide a full example so we can debug? Thanks in advance.

No. It's just Chisel 3.2 which the latest version. 
> """ ;buildInfoPackage: chisel3, version: 3.2-SNAPSHOT, scalaVersion: 2.12.6, sbtVersion: 1.1.1 """

edwardcwang, 2018-12-25:
Just tried it on chisel3 and firrtl master, and it appears to work and generate Verilog:

```verilog
module TestMod(
  input         clock,
  input         reset,
  input         io_pad_fillHead,
  input         io_pad_fillTail,
  input         io_ddr_weight_en,
  output [14:0] io_addr_out
);
  reg [14:0] addr_gen; // @[<console> 24:23]
  reg [31:0] _RAND_0;
  wire [14:0] _T_3; // @[<console> 30:26]
  assign _T_3 = addr_gen + 15'h1; // @[<console> 30:26]
  assign io_addr_out = addr_gen; // @[<console> 33:13]
`ifdef RANDOMIZE_GARBAGE_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_INVALID_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_REG_INIT
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_MEM_INIT
`define RANDOMIZE
`endif
`ifndef RANDOM
`define RANDOM $random
`endif
`ifdef RANDOMIZE
  integer initvar;
  initial begin
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif
  `ifdef RANDOMIZE_REG_INIT
  _RAND_0 = {1{`RANDOM}};
  addr_gen = _RAND_0[14:0];
  `endif // RANDOMIZE_REG_INIT
  end
`endif // RANDOMIZE
  always @(posedge clock) begin
    if (reset) begin
      addr_gen <= 15'h0;
    end else begin
      if (io_pad_fillHead) begin
        addr_gen <= 15'h0;
      end else begin
        if (io_pad_fillTail) begin
          addr_gen <= 15'h4000;
        end else begin
          if (io_ddr_weight_en) begin
            addr_gen <= _T_3;
          end
        end
      end
    end
  end
endmodule
```