chipsalliance/firrtl #986: RFC: Annotating DoPrims, Muxes, and ValidIfs

azidar, 2019-01-02:
**Feature Request**: Enabling annotating non-named nodes through unique tags

**What is the use case for changing the behavior?**
Right now, you cannot track non-named FIRRTL nodes such as a mux. The current work-around is to make sure that the mux is not in a nested expression, and then you annotate the name of the signal that is assigned its output. While this works generally, its not a robust solution. If the mux is inlined, the signal is deleted, but the mux is still around.

In addition, we are using the [[Target]] classes to build connectivity graphs (see #937), where these non-named nodes play an important role (and should be represented explicitly). The work-around for that PR is to make [[ReferenceTarget]]s with special names, where the first mux in the module is named "@mux#0", where the 0 is a unique module-scoped tag to single out a particular mux. This works well within a transform, but this mechanism to identify the mux is not robust under arbitrary transformations, and thus should not leave the transform.

Unfortunately, you often want this information to leave the transform. When recording the clock source of a given signal, the clock source may be an `asClock` non-named expression. We have no way to robustly record this information to live outside the transform. (A potential solution could be to transform the circuit to give the `asClock` a unique reference, e.g. pull it into its own module. This does seem heavy-handed, but I'm open to discussing it).

**Impact**: API modification? API addition?

**Development Phase**: Proposal

My current proposal is to add an optional tag argument (`#tag`) to every non-named node (DoPrims, Mux, ValidIf): e.g. `add(x, y, #1)`, `mux(x, y, z, #0)`

The tag can be any positive integer, but must be unique among tags of other non-named nodes of its type. There can be an add with tag 0 and a mux with tag 0, but cannot have two muxes with tag 0.

When creating a new non-named node, you do not have to provide it with a tag. Only when you want a tag, can you add it (in which case we will provide library functions for guaranteeing a unique tag). This will preserve backwards compatibility. During a check pass (like with ChieckHighForm), we will ensure that no two non-named nodes of the same type share a tag.

Mappers will always preserve the tag argument. If you are changing DoPrim, you should preserve the tag (e.g. use expr.copy).

We will provide a corresponding [[Target]] class which will reference these unnamed nodes, and then you can update the RenameMap with any changes in your transform to these nodes.

Thoughts?

Questions to seed discussion:

- Would you find annotating unnamed signals/components useful?
- Should we add tags to Connect, PartialConnect, IsInvalid?
- What about Print, Stop, Conditionally? Types? Literals?
- Do you perceive any backwards-incompatibilities that I haven't thought of?

davidbiancolin, 2019-01-03:
> Should we add tags to Connect, PartialConnect, IsInvalid?
> What about Print, Stop, Conditionally? Types? Literals?

Is there any reason not to support these? I'm a proponent of being able to annotate any node. MIDAS already wants to be able to annotate prints, and i can see use cases for Stop.

azidar, 2019-01-04:
I suppose the reason to not support some are that some could have difficult semantics to define.

I think supporting Print, Stop, and Literals are no different than DoPrims etc. 

Tagging Conditionally would require some thinking, but my gut feeling is it wouldn't be too bad, we'd just add a tag argument to the when clause.

I'm not sure how type annotations would work, but if you can annotate non-named signals, then I suppose you can always just reference the object itself and implicitly imply the type?

Connect/PartialConnect/IsInvalid would be a bit tougher to define the semantics of. Some example scenarios would be its behavior under ExpandWhens or Wire->Node conversions.

jackkoenig, 2019-01-04:
> The tag can be any positive integer, but must be unique among tags of other non-named nodes of its type. There can be an add with tag 0 and a mux with tag 0, but cannot have two muxes with tag 0.

What is the motivation for different types of nodes to have the same tags?

Similarly to how names are not scoped by their type (we just have `a`, we don't have `node#a` and `reg#a`), I would think we should just have a single tag namespace per module. My argument is mainly that it feels simpler and less error prone.

azidar, 2019-01-04:
> What is the motivation for different types of nodes to have the same tags?

Yeah, the reason is it makes name changes more robust to small changes in the compiler. `mux#30` is the 30th mux in the design, rather than the 30th non-named thing in the design.