chipsalliance/firrtl #988: UniquifySpec rename deep bundles isn't quick under coverage tests

ucbjrl, 2019-01-04:
**Type of issue**: other enhancement

  * **What is the current behavior?**
```bash
sbt ++2.12.4 clean coverage test coverageReport
 ...
[info] UniquifySpec:
[info] Uniquify
 ...
[info] - should quickly rename deep bundles *** FAILED ***
[info]   false was not equal to true (UniquifySpec.scala:303) 
```
 * **What is the expected behavior?**
  * **Please tell us about your environment:**
    - version: `1.2-SNAPSHOT`
    - OS: `Linux knight 4.4.0-134-generic #160-Ubuntu SMP Wed Aug 15 14:58:00 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`
    - OS: `Darwin Kernel Version 18.2.0: Mon Nov 12 20:24:46 PST 2018; root:xnu-4903.231.4~2/RELEASE_X86_64 x86_64`

**What is the use case for changing the behavior?**
coverage test should pass

seldridge, 2019-01-04:
FWIW, I've gotten this to fail under normal, non-coverage tests on one server. I couldn't think of a good solution, though.

ucbjrl, 2019-01-04:
I think the _correct_ solution is to count bytecode executed in both the `depth = 0` and `depth = 500` cases and verify that the latter is a reasonable multiple of the former. I started down that road until I discovered there's no Java API to extract performance information from the JVM. You need to write an agent in C, which introduces portability issues.

I assume someone will eventually develop a Java API for performance analysis, at which time, we should re-visit this.

ucbjrl, 2019-02-21:
This test still occasionally fails:
```
[info]   9038.99351 was not less than 6113.865 (UniquifySpec.scala:317)
```

closed 2019-05-03 by azidar