chipsalliance/firrtl #998: instance different verilog module when constant propagation.

sequencer, 2019-01-22:
**Type of issue**: other enhancement

**What is the use case for changing the behavior?**
1. optimize the constant propagating behavior by replace xor(a,1) to not(a) etc,
2. instance different verilog module when some ports are tie 0/1 rather than single instance with constant as input.    

**Impact**: API modification
**Development Phase**: request

Here is the example code by chisel3
```
import chisel3._

trait ConstantPropagationTest {
  class CSAOutBundle extends Bundle {
    val s: UInt = UInt(32.W)
    val c: UInt = UInt(32.W)
  }

  def csa(a: UInt, b: UInt, c: UInt): CSAOutBundle = {
    // the reason I use Module here is I wanna keep the circuit hierarchy. 
    val m = Module(new Module {
      val io = IO(new Bundle {
        val a = Input(UInt(32.W))
        val b = Input(UInt(32.W))
        val c = Input(UInt(32.W))
        val out = Output(new CSAOutBundle)
      })
      val a = Wire(Vec(32, Bool()))
      val b = Wire(Vec(32, Bool()))
      val c = Wire(Vec(32, Bool()))
      val cOut = Wire(Vec(32, Bool()))
      val sOut = Wire(Vec(32, Bool()))
      a := io.a.asTypeOf(Vec(32, Bool()))
      b := io.b.asTypeOf(Vec(32, Bool()))
      c := io.c.asTypeOf(Vec(32, Bool()))
      io.out.s := sOut.asTypeOf(UInt(32.W))
      io.out.c := cOut.asTypeOf(UInt(32.W))
      override def desiredName = "csa"
      for(i <- 0 to 31) {
        sOut(i) := a(i) ^ b(i) ^ c(i)
        cOut(i) := {
          i match {
            case j if j == 0 => false.B
            case default => (a(default - 1) && b(default - 1)) || (b(default - 1) && c(default - 1)) || (a(default - 1) && c(default - 1))
          }
        }
      }
    })
    m.io.a := a
    m.io.b := b
    m.io.c := c
    m.io.out
  }
}

class ConstantTester(dual: Boolean) extends Module
  with ConstantPropagationTest {
  val io = IO(new Bundle {
    val a = Input(UInt(32.W))
    val b = Input(UInt(32.W))
    val out0 = Output(new CSAOutBundle)
    val out1 = Output(new CSAOutBundle)
  })
  val c = WireInit("h1234abcd".U)
  io.out0 := csa(io.a, io.b, c)
  io.out1 := {
    dual match {
      case i if i => csa(c, io.a, io.b)
      case i if !i => DontCare
  }}
}

object ConstantTesterEmitter extends App {
  chisel3.Driver.execute(
    Array("-td", "./constantTesterDualInstance",
    ),
    () => new ConstantTester(true))
  chisel3.Driver.execute(
    Array("-td", "./constantTesterSingleInstance",
    ),
    () => new ConstantTester(false))
}

```
while propagating a `"h1234abcd".U` to one csa, the emitted verilog will just omit the constant port, and tie constant inside verilog(port still exist in firrtl).
However the while instance two modules, these enhancement won't exist, the emitted verilog will only have one module, and propagating the constant to port.
And by the way the xor constant optimize will give a logic like this:
```
assign _T_730 = a_7 ^ b_7; // @[PlayGround.scala 29:25]
assign sOut_7 = _T_730 ^ 1'h1; // @[PlayGround.scala 29:32]
```
obviously, `sOut_7 = ! _T_730`

jackkoenig, 2019-01-22:
Thanks for the heads up about `xor(a, 1)` => `not(a)`

Per the cross-module optimizations--it is intended behavior that we preserve the module structure and don't optimize across inputs if they have different values driving them. Preserving these modules is critical for hierarchical place-and-route flows. The [Hart ID](https://github.com/freechipsproject/rocket-chip/blob/4f0ca76fa52b0191ccb0a90194cbcf0b3d60dac0/src/main/scala/subsystem/RocketSubsystem.scala#L74) in rocket-chip is an example of something that we could optimize in this way but intentionally do not.

Can you provide more information about why you would prefer the modules be duplicated and optimized separately rather than preserving structure? FIRRTL's optimizations are not intended to be a complete suite that results in the minimal amount of hardware--commercial and open-source synthesis tools do a much more thorough job than we ever intend to. Rather, the optimizations are intended to clean up the output of Chisel generators to make the resulting Verilog more amenable to existing flows.

sequencer, 2019-01-22:
Because I wanna custom these generated module with combinational logic gate(rather than flat and send into backend synthesis), using which as a hard IP to replace the original module.
If we use the standard backend method, I have to flat my circuit and generate a optimized circuit, in such case, without the hierarchy, I can't find which circuit I should replace.

sequencer, 2019-01-22:
And there might be another problem with these situation:
```
  def fullAdder(a: UInt, b: UInt, name: String): UInt = {
    val m = Module(new Module {
      val io = IO(new Bundle {
        val a = Input(UInt(32.W))
        val b = Input(UInt(32.W))
        val out = Output(UInt(32.W))
      })
      override def desiredName = s"adder_$name"
      io.out := io.a + io.b
    })
    m.io.a := a
    m.io.b := b
    m.io.out
  }

class AdderTester extends Module
  with ConstantPropagationTest {
  val io = IO(new Bundle {
    val a = Input(UInt(32.W))
    val b = Input(UInt(32.W))
    val out = Output(Vec(2, UInt(32.W)))
  })

  io.out(0) := fullAdder(io.a, io.b, "fast")
  io.out(1) := fullAdder(io.a, io.b, "slow")
}
```
this will lead to output verilog has a `adder_null` module name

sequencer, 2019-01-25:
workarounded by adding `NoDedupAnnotation` annotation.

closed 2019-01-25 by sequencer

jackkoenig, 2019-01-25:
Ah yes I think that is the right solution; glad you were able to resolve the issue!

sequencer, 2019-06-24:
I talk about this issue to @edwardcwang last week.
Some `ConstantPropagation` chance might be missed by of `Dedup`.
I think there might be some way to elegantly resolve this by annotating modules with some specific `Annotation`.
`doNotDedup` will deliver too much identical modules, but in this case, we just need different modules with different constant propagation result.

jackkoenig, 2019-06-25:
The issue is that this is the desired behavior. Missing out on some amount of constant propagation is nothing compared to the benefits of hierarchical place and route.

Can you elaborate a bit on how big of an issue this is for you? Generally the FIRRTL optimizations are intended to improve performance of downstream simulators and synthesis tools, as well as improve integration with existing tools to avoid common lint errors. We don't pretend to provide full optimization functionality, leaving that to the actual synthesis tools. Is there a particular tool or flow you're having trouble integrating with?