chipsalliance/firrtl #1020: Flatten transform generates not-very-user-friendly Module names.

ccelio, 2019-02-21:
**Type of issue**:feature request

  * **What is the current behavior?**

If I make a trait as follows:
```scala
trait FlattenInstance { self: chisel3.experimental.BaseModule =>
  chisel3.experimental.annotate(
    new chisel3.experimental.ChiselAnnotation {
      def toFirrtl: firrtl.annotations.Annotation = firrtl.transforms.FlattenAnnotation(self.toNamed) } )
}
``` 

And then use it as follows:

```scala
class myModuleA extends Module {
  val b = Module(new myModuleB extends FlattenInstance)
}
```

The resulting Verilog yields:

```verilog
myModuleAanon1 b(...)
```

  * **What is the expected behavior?**

```verilog
myModuleB b(...)
```

The current behavior is not very user-friendly, especially for handing off the Verilog to another team; the current work-around is to manually add `with InlineInstance` to all submodules which yields the desired behavior.

  * **Please tell us about your environment:**
    - firrtl version: 50ba95fe05cc348496425733554be5fc7d797de1
    - chisel3 version: https://github.com/freechipsproject/chisel3/commit/9a0ce2272c9d5d0a8bdc90e84269749ce054664d

**Impact**: no functional change

seldridge, 2019-02-21:
The difficulty is that this is an anonymous module and it really isn't a `myModuleB` anymore -- it's some anonymous child created inside of `myModuleA`. This is one issue that didn't occur to me with the `FlattenInstance` trait API as opposed to using a more explicit annotator method which wouldn't have this problem.

This is annoying because I like the mixin API, but this is creating anonymous children. I think this may motivate a better `desiredName` API that will generate names based on the mixins (or optionally, keep the name of the parent). Obviously, this turns into a mess with multiple mixins, but I think we can think of something here.

In the meantime, how about overriding `desiredName`? Something like the following should do the trick:

```scala
val b = Module(new myModuleB extends FlattenInstance { override val desiredName = "myModuleB" })
```

ccelio, 2019-02-21:
The recommended solution of overriding desiredName appears to work, which is an acceptable solution.

Feel free to close if desired.