chipsalliance/firrtl #1021: SyncReadMem should generate a register to register data rather than address

MikuChaser, 2019-02-21:
- Type of issue: other enhancement
- version: `3.2-SNAPSHOT`
- OS: `Ubuntu 16.04`

I want to create a true double-port RAM in Chisel and synthesis the Verilog code in Vivado 2018.3. Here is my Chisel code:
`class DoublePortsRAM extends Module {
  val io = IO(new Bundle { 
    val addr1 = Input(UInt(10.W))
    val dataIn1 = Input(UInt(32.W))
    val en1 = Input(Bool())
    val we1 = Input(Bool())
    val dataOut1 = Output(UInt(32.W))
    val clk2 = Input(Clock())
    val reset2 = Input(Bool())
    val addr2 = Input(UInt(10.W))
    val dataIn2 = Input(UInt(32.W))
    val en2 = Input(Bool())
    val we2 = Input(Bool())
    val dataOut2 = Output(UInt(32.W))  
  })

  val syncRAM = SyncReadMem(1024, UInt(32.W))

  when(io.en1) {
    when(io.we1) {
      syncRAM.write(io.addr1, io.dataIn1)
      io.dataOut1 := DontCare
    } .otherwise {
      io.dataOut1 := syncRAM.read(io.addr1)
    }
  } .otherwise {
    io.dataOut1 := DontCare
  }

  withClockAndReset(io.clk2, io.reset2) {
    when(io.en2) {
      when(io.we2) {
        syncRAM.write(io.addr2, io.dataIn2)
        io.dataOut2 := DontCare
      } .otherwise {
        io.dataOut2 := syncRAM.read(io.addr2)
      }
    } .otherwise {
      io.dataOut2 := DontCare
    }
  }
}`
After compiling it, the Verilog code of reading data like these:
`always @(posedge clock) begin
    ...
    if (_GEN_36) begin
      syncRAM__T_1_addr_pipe_0 <= io_addr1;
    end
  end
...
  assign syncRAM__T_1_addr = syncRAM__T_1_addr_pipe_0;
  assign syncRAM__T_1_data = syncRAM[syncRAM__T_1_addr]; 
  assign io_dataOut1 = syncRAM__T_1_data;
...`
However, the Vivado reports that it can't infer the pattern of the RAM. I must modify the Verilog code manually to these:
`always @(posedge clock) begin
    ...
    if (_GEN_36) begin
      syncRAM__T_1_addr_pipe_0 <= syncRAM[io_addr1];
    end
  end
... 
  assign io_dataOut1 = syncRAM__T_1_addr_pipe_0;
...`
In other words, the Synchronous Read RAM shouldn't use a register to store the reading address. On the contrary, it is OK to store the reading data. In addition, Vivado can synthesis the single-port RAM without modifying the Verilog. What's wrong with it?

I think we just need to change the behavior of reading data when generating FIRRTL. I have read the Verilog templates of RAM that provided by Vivado, and all of them use a register to store reading data rather than reading address. 

Impact: no functional change

Development Phase: request

aswaterman, 2019-02-21:
For situations where SyncReadMems get mapped to flops, this change would have a significant impact on how pipeline stages are balanced (moving all of the read delay onto the address side).  That has the potential to be a disruptive change for some users.

You might consider using the ReplSeqMem pass so you can emit custom Verilog that's Vivado-friendly, as we do here: https://github.com/freechipsproject/rocket-chip/blob/master/scripts/vlsi_mem_gen

MikuChaser, 2019-02-22:
> For situations where SyncReadMems get mapped to flops, this change would have a significant impact on how pipeline stages are balanced (moving all of the read delay onto the address side). That has [...]

Do you mean that I need a custom script to emit custom Verilog?

albert-magyar, 2019-11-05:
This is now supported as the Verilog emission pattern when the `read-under-write` parameter is explicitly set to `read-first`.

albert-magyar, 2019-11-05:
Supported in #1183

closed 2019-11-05 by albert-magyar