chipsalliance/firrtl #1022: FIRRTL is happy to rename main when deduping

ucbjrl, 2019-02-21:
**Type of issue**: bug report

The `main` module should never be a candidate for renaming.

The following FIRRTL:
```
circuit main:
  module dupe:
    input in: UInt<8>
    output out: UInt<8>
    out <= in
  module main:
    input in: UInt<8>
    output out: UInt<8>
    out <= in
```
will die with the following:
```bash
Exception in thread "main" firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
	at firrtl.Utils$.error(Utils.scala:440)
	at firrtl.Utils$.throwInternalError(Utils.scala:181)
	at firrtl.Driver$.$anonfun$execute$1(Driver.scala:252)
	at logger.Logger$.$anonfun$makeScope$2(Logger.scala:138)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at logger.Logger$.makeScope(Logger.scala:136)
	at firrtl.Driver$.execute(Driver.scala:224)
	at firrtl.Driver$.execute(Driver.scala:302)
	at firrtl.Driver$.main(Driver.scala:318)
	at firrtl.Driver.main(Driver.scala)
Caused by: java.util.NoSuchElementException: key not found: main
	at scala.collection.MapLike.default(MapLike.scala:231)
	at scala.collection.MapLike.default$(MapLike.scala:230)
	at scala.collection.AbstractMap.default(Map.scala:59)
	at scala.collection.mutable.HashMap.apply(HashMap.scala:65)
	at firrtl.transforms.CheckCombLoops.run(CheckCombLoops.scala:258)
 ...
```

  * **What is the current behavior?**
If `main` is the last of a set of duplicate modules, it will be removed and FIRRTL references to it replaced with its surviving duplicate. However, the FIRRTL compiler has squirreled away references to `main` and it will be upset when it disappears.

  * **What is the expected behavior?**
`main` should win any deduplication battles.

  * **Please tell us about your environment:**
- Darwin dhcp-35-82.EECS.Berkeley.EDU 18.2.0 Darwin Kernel Version 18.2.0: Thu Dec 20 20:46:53 PST 2018; root:xnu-4903.241.1~1/RELEASE_X86_64 x86_64
- FIRRTL afdb780aeca80c1fed94dd20fd22037490169472

**What is the use case for changing the behavior?**
Preserve the `main`!

**Impact**: no functional change

**Development Phase**: proposal

chick, 2019-02-21:
I think the use case you are looking for is "Remember the Main!"

ucbjrl, 2019-02-21:
Missed it by that much ...

seldridge, 2019-02-21:
Fixed by #1023

closed 2019-02-21 by seldridge