chipsalliance/firrtl #1042: when i run "sbt compile", error happened:unknown tag html in file ivy-5.0.3.xml.original

botaichang, 2019-02-28:
when i run "sbt compile", error happened as follows, could anyone help me please?
```bash
java.text.ParseException: [xml parsing: ivy-5.0.3.xml.original:2:7: cvc-elt.1: Cannot find the declaration of element 'html'. in file:~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original
[error]unknown tag html in file:~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original
[error]unknown tag head in file:~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original
[error]unknown tag meta in file:~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original
[error]unknown tag meta in file:~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original
[error]unknown tag meta in file:~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original
[error]unknown tag meta in file:~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original
[error]unknown tag title in file:~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original
[error]unknown tag link in file:~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original
[error]unknown tag link in file:~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original
[Fatal Error] ivy-5.0.3.xml.original:11:3: The element type "meta" must be terminated by the matching end-tag "</meta>". 
```

azidar, 2019-02-28:
Did you try deleting your ivy2 cache and see what happens?

botaichang, 2019-03-01:
> Did you try deleting your ivy2 cache and see what happens?

@azidar Hi azidar,Thank you very much for reply. I have tried deleting the cache file and it gives the same error.

grebe, 2019-03-01:
I don't think firrtl depends on (even transitively) on that library (asm). Are you using a special branch or other dependencies? I used that library for a branch of treadle.

botaichang, 2019-03-01:
> I don't think firrtl depends on (even transitively) on that library (asm). Are you using a special branch or other dependencies? I used that library for a branch of treadle.

Hi grebe,Thanks for reply.  I am using the master branch of firrtl.

grebe, 2019-03-01:
Then I'm having trouble seeing how this could be related to firrtl. Do you have any global sbt plugins? Maybe try blowing away the ivy2 folder again and all the target/ directories?

ucbjrl, 2019-03-01:
I believe this is a bug in one of `sbt`'s components, and depends on which version of `sbt` was active when the cache was populated with this item. I suspect the dependency is due to `sbt` itself - the file `~/.ivy2/cache/org.ow2.asm/asm/ivy-5.0.3.xml.original` is in my `~/.ivy2/cache, but it dates from May 24, 2014 and doesn't contain any of the unknown tags in your report.

I would try deleting the file (and any other `.xml.original` files that generate errors), or as @azidar and @grebe suggest, delete your `~/.ivy2/cache` directory. It's a good idea to do this periodically anyway to eliminate old versions of components that are no longer required.

botaichang, 2019-03-04:
@azidar @grebe @ucbjrl  I delete the .ivy2/ directory and target/ directory,and it gets unknown tags in many .xml.original files.   And there is no global sbt plugin here. 
```bash
java.text.ParseException: [xml parsing: ivy-9.xml.original:2:7: cvc-elt.1: Cannot find the declaration of element 'html'. in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
, unknown tag html in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
, unknown tag head in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
, unknown tag title in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
, unknown tag link in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
, unknown tag link in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
, [Fatal Error] ivy-9.xml.original:11:3: The element type "meta" must be terminated by the matching end-tag "</meta>". in file:~/.ivy2/cache/org.sonatype.oss/oss-parent/ivy-9.xml.original
]
	java.lang.NullPointerException
	java.lang.NullPointerException
	java.text.ParseException: [xml parsing: ivy-2.14.6.xml.original:2:7: cvc-elt.1: Cannot find the declaration of element 'html'. in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
, unknown tag html in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
, unknown tag head in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
, unknown tag meta in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
, unknown tag meta in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
, unknown tag meta in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
, unknown tag meta in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
, unknown tag title in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
, unknown tag link in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
, unknown tag link in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
, [Fatal Error] ivy-2.14.6.xml.original:11:3: The element type "meta" must be terminated by the matching end-tag "</meta>". in file:~/.ivy2/cache/jline/jline/ivy-2.14.6.xml.original
]
	java.lang.NullPointerException
	java.lang.NullPointerException

		:: org.apache.logging.log4j#log4j-api;2.11.1: several problems occurred while resolving dependency: org.apache.logging.log4j#log4j-api;2.11.1 {compile=[compile(*), master(*)], runtime=[runtime(*)]}:
	several problems occurred while resolving dependency: org.apache.logging.log4j#log4j;2.11.1 {}:
	several problems occurred while resolving dependency: org.apache.logging#logging-parent;1 {}:
	several problems occurred while resolving dependency: org.apache#apache;18 {}:
	java.lang.NullPointerException
	java.text.ParseException: [xml parsing: ivy-18.xml.original:2:7: cvc-elt.1: Cannot find the declaration of element 'html'. in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
, unknown tag html in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
, unknown tag head in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
, unknown tag title in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
, unknown tag link in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
, unknown tag link in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
, [Fatal Error] ivy-18.xml.original:11:3: The element type "meta" must be terminated by the matching end-tag "</meta>". in file:~/.ivy2/cache/org.apache/apache/ivy-18.xml.original
]
	java.lang.NullPointerException
	java.lang.NullPointerException
	java.text.ParseException: [xml parsing: ivy-1.xml.original:2:7: cvc-elt.1: Cannot find the declaration of element 'html'. in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
, unknown tag html in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
, unknown tag head in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
, unknown tag meta in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
, unknown tag title in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
, unknown tag link in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
, unknown tag link in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
, [Fatal Error] ivy-1.xml.original:11:3: The element type "meta" must be terminated by the matching end-tag "</meta>". in file:~/.ivy2/cache/org.apache.logging/logging-parent/ivy-1.xml.original
]
	java.lang.NullPointerException
```

botaichang, 2019-03-04:
would it be possible to package all dependency libraries into one big tar file, so that we can install the software tools more conveniently?

grebe, 2019-03-04:
I'm curious what system you're on and what versions of sbt + java. These are very strange errors that seem to indicate something flaky is going on. Are you low on disk space or have a flaky network connection? I'm not finding much meaningful information out there about these errors.

We don't package things individually- there are different components that people sometimes use and sometimes don't, and generally sbt is pretty reliable in figuring that out and fetching dependencies. The closest thing I can think of is the coursier tool, which resolves dependencies and downloads them for you without using sbt. After install coursier, the command
```
coursier fetch -p -r sonatype:snapshots edu.berkeley.cs:chisel3_2.12:3.2-SNAPSHOT
```

should fetch all the files you need to use chisel and then print (because of the `-p` flag) a classpath that you can use to build scala programs using all the dependencies. I'm not sure how to get it to make a single fat jar, though.

botaichang, 2019-03-04:
> I'm curious what system you're on and what versions of sbt + java. These are very strange errors that seem to indicate something flaky is going on. Are you low on disk space or have a flaky network [...]

@grebe Hi grebe, thanks for reply. my environment is as follows:
- system:  ubuntu 16.04
- sbt: 1.1.1
- scala: 2.12.7
- java: 1.8.0

botaichang, 2019-03-04:
@grebe @azidar @ucbjrl  I found that the error file format is different from other correct file's. 
the right file format is this:(take ivy-6.xml.original as an example) 
```bash
<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 
    <modelVersion>4.0.0</modelVersion>
    <groupId>org.sonatype.oss</groupId>
    <artifactId>oss-parent</artifactId>
    <version>6</version>
    <packaging>pom</packaging>
    <name>Sonatype OSS Parent</name>
    <url>http://nexus.sonatype.org/oss-repository-hosting.html</url>
    <description>Sonatype helps open source projects to set up Maven repositories on https://oss.sonatype.org/ </description>
    <scm>
        <connection>scm:svn:http://svn.sonatype.org/spice/tags/oss-parent-6</connection>
        <developerConnection>scm:svn:https://svn.sonatype.org/spice/tags/oss-parent-6</developerConnection>
        <url>http://svn.sonatype.org/spice/tags/oss-parent-6</url>
    </scm>
```
But the reported error file format is this(take ivy-9.xml.original  as an example):
```bash
<html>
<head>
<meta content='text/html; charset=UTF-8' http-equiv='Content-Type'>
<meta content='3CbaVvw-I7MlrmmmHz0bfbko7oMCW1mn2u65uWsWWB8' name='google-site-verification'>
<meta content='width=device-width, initial-scale=1.0' name='viewport'>
<meta content='telephone=no' name='format-detection'>
<title>reficio.org is expired</title>
<link rel="stylesheet" media="screen" href="https://fonts.googleapis.com/css?family=Open+Sans:300,400,600,700" />
<link rel="stylesheet" media="all" href="/assets/application-2f7e7f30d812d0f3950918c7562df7e68eeeebd8649bdea2bc3844eb07fc8269.css" />
</head>
<body>
<header>
<a rel="nofollow" href="https://www.hover.com/?source=expired"><img width="102" height="30" src="/assets/hv_logo_retina-6a2ba8350907d4a17bfc7863c2f1378e38a53bd22b790c69c141
</header>
<main>
<h1>reficio.org</h1>
<h2>has expired.</h2>
...
```

edwardcwang, 2019-03-04:
Re "would it be possible to package all dependency libraries into one big tar file" -> another possible option is to build a "fat JAR", though that typically would also include FIRRTL itself.

botaichang, 2019-03-05:
@grebe @azidar @ucbjrl  @edwardcwang  Hi , thanks all for kindly help. I am now can pass this kind of error by deleting many cache files including the .ivy2 and .m2 files. And modify my maven configurations.Thanks very much.   And there is another problem when i compile,  here is the error:
```bash
build.sbt:5: error: not found: value SiteScaladocPlugin
enablePlugins(SiteScaladocPlugin)
```

botaichang, 2019-03-05:
@edwardcwang  Yes, I think it's a good idea to build a "fat JAR" so that we can only download one big whole file for use. It's a little pain to download all the dependencies by sbt for some network reason for example network firewalls in some places.