chipsalliance/firrtl #1044: AsyncReset literals check is NOT compatible with chisel's complex literals emission

johnsbrew, 2019-03-04:
**Type of issue**: bug report

**Steps to reproduce the bug:**
Compile the following fragment
`./firrtl -i bug_asyncreset.fir -tn Test`
  * **Relevant FIRRTL fragment:**
```
circuit Test : 
  module Test : 
    input clock : Clock
    input reset : AsyncReset
    @[NB: following lines were generated by chisel]
    @[val valid = RegInit(VecInit(Seq.fill(8)(false.B)))]
    wire _T_6 : UInt<1>[8]
    _T_6[0] <= UInt<1>("h00")
    _T_6[1] <= UInt<1>("h00")
    _T_6[2] <= UInt<1>("h00")
    _T_6[3] <= UInt<1>("h00")
    _T_6[4] <= UInt<1>("h00")
    _T_6[5] <= UInt<1>("h00")
    _T_6[6] <= UInt<1>("h00")
    _T_6[7] <= UInt<1>("h00")
    reg valid : UInt<1>[8], clock with : (reset => (reset, _T_6))
```
  * **What is the current behavior?**
```
Exception in thread "main" firrtl.passes.CheckHighForm$NonLiteralAsyncResetValueException: 
@[matching_fifo.scala 57:37:bug_asyncreset.fir@15.4]: 
[module Test] AsyncReset Reg 'valid' reset to non-literal '_T_6'
```
  * **What is the expected behavior?**
No errors: _T_6 is actually a literal

  * **Environment:**
    - version: `3.2-SNAPSHOT`
    - OS: `Darwin 17.7.0 Darwin Kernel Version 17.7.0: Thu Dec 20 21:47:19 PST 2018; root:xnu-4570.71.22~1/RELEASE_X86_64 x86_64`

**What is the use case for changing the behavior?**
Ensuring chisel's firrtl emission compatibility, see the simple example given
`val valid = RegInit(VecInit(Seq.fill(8)(false.B)))`
**Impact**: no functional change

**Development Phase**: request 

**Other information**

edwardcwang, 2019-03-04:
Part of the issue could be that VecInits aren't real literals in Chisel - they are wires that get assigned to, as opposed to "real" literals. See https://github.com/freechipsproject/chisel3/issues/849

johnsbrew, 2019-03-04:
@edwardcwang I understand your point about Vec but it not really the issue here since some other chisel constructs could potentially lead to the same behaviour.
To fix this issue in a generic way I propose a recursive lookup, see #1045

jackkoenig, 2019-03-04:
@johnsbrew thanks for the report and proposed fix. I was hoping to avoid to having to do such a recursive check, but in the absence of support for aggregate literals, I think such an approach is necessary.

edwardcwang, 2019-03-04:
What's the status of bundle/aggregate literals in FIRRTL? That could be another way (long term maybe) to avoid this recursive check.