chipsalliance/firrtl #1047: Legal Chisel 2 code breaks

schoeberl, 2019-03-06:
**Type of issue**: bug report

Following Chisel 2 code:

```
  val rdAddr = Wire(UInt())

  // default
  rdAddr := Cat(UInt(0, 1), nextPC)
  switch(stateReg) {
    is(fetch) {
...
      when(rdData(7) === Bits(0)) {
        rdAddr(8, 4) := UInt(0x10)
        rdAddr(3, 0) := rdData
```

breaks with 
firrtl.passes.CheckGenders$WrongGender:  @[Lipsi.scala 127:22:@125.8]: [module Lipsi]  Expression _T_74 is used as a FEMALE but can only be used as a MALE.
[error] firrtl.passes.CheckGenders$WrongGender:  @[Lipsi.scala 128:22:@127.8]: [module Lipsi]  Expression _T_76 is used as a FEMALE but can only be used as a MALE.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**
        node _T_74 = bits(rdAddr, 8, 4) @[Lipsi.scala 127:15]
        _T_74 <= UInt<5>("h010") @[Lipsi.scala 127:22]
        node _T_76 = bits(rdAddr, 3, 0) @[Lipsi.scala 128:15]
        _T_76 <= mem.io.rdData @[Lipsi.scala 128:22]

  * **Please tell us about your environment:**
    (examples)
    - version: `3.1.2`
    - OS: macOS

**What is the use case for changing the behavior?**

Porting Chisel 2 code to Chisel 3

edwardcwang, 2019-03-06:
> `rdAddr(8, 4) := UInt(0x10)`

I think the issue here is that subwork assignment is illegal in Chisel3/FIRRTL, no?

schoeberl, 2019-03-06:
> On 6 Mar, 2019, at 21:39, edwardcwang <notifications@github.com> wrote:
> 
> rdAddr(8, 4) := UInt(0x10)
> 
> I think the issue here is that subwork assignment is illegal in Chisel3/FIRRTL, no?
> 
> 

Ok, but this was legal Chisel 2 code.

Martin

edwardcwang, 2019-03-06:
Did chisel2 ever work with FIRRTL though? I thought chisel2 generated its Verilog directly as this was pre-FIRRTL days. Chisel3 has a "Chisel2 compatibility layer" but I'm not sure they can handle this case, which is why I CC'ed Richard above.

schoeberl, 2019-03-06:
This does not really matter when one ports Chisel 2 code to Chisel 3. It breaks. I am using the Chisel 2 compatibility mode.

edwardcwang, 2019-03-06:
In that case, let's ask @ducky64 for his opinion on how the Chisel 2 compatibility mode should handle this.

ducky64, 2019-03-06:
chisel2 compatibility mode isn't perfect (and I don't think this is the only example of code that breaks), though we typically try to at least generate informative error messages.

That this isn't generating a good diagnostic error (something more like "subword assignment not supported, you should do X or Y instead, see Z for details") is potentially a bug. The chisel3 binding system should catch this before it gets to FIRRTL.

There's more context in freechipsproject/chisel3#244 and freechipsproject/chisel3#878

We can always discuss adding back this feature, though all the previous rounds of discussion on that have led to a resounding no.

schoeberl, 2019-03-06:
Is it really a compatibility mode issue? With Chisel 3 I get following error:

[error] (run-main-0) chisel3.internal.ChiselException: Cannot reassign to read-only chisel3.core.UInt@d6

Does this mean that it has been decided that Chisel 2 subword assignment is not supported in Chisel 3? If so, we can simply close the issue.

Although, this is not a so uncommon pattern in hardware description.

Cheers,
Martin

ducky64, 2019-03-06:
Yes, that is the case - subword assignment is not supported in Chisel3. Issues freechipsproject/chisel3#244 and freechipsproject/chisel3#878 contain more discussion and rationale.

If you've got new use cases that have not been brought up before, we can certainly discuss them, but you should also be aware of the prior conversations including suggested alternate paths.

schoeberl, 2019-03-07:
Thanks for the pointers. Looks like I am stumbling over this not the first time ;-)

closed 2019-03-07 by schoeberl