chipsalliance/firrtl #1048: Error in testing mode

schoeberl, 2019-03-07:
**Type of issue**: bug report

The issue can be reproduced
  * **What is the current behavior?**
 Error
  * **What is the expected behavior?**
Compile (as it did with Chisel 2)

Can be reproduced with:
Clone https://github.com/schoeberl/lipsi
Checkout branch chisel3 
A simple “make" will produce the error

The error log is:

[error] (run-main-0) firrtl.passes.PassExceptions: 
[error] firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Conditional.scala 37:30:@102.4]: [module Lipsi] Reference stateReg is not declared.

ucbjrl, 2019-03-08:
This appears to be due to a bug in the FIRRTL `RemoveWires` transform. The following STC:
```scala
import chisel3._

class RefBeforeDef extends Module {
  val io = IO(new Bundle {
    val out = Output(UInt(8.W))
  })

  val out = Wire(UInt(8.W))
  out := 0.U
  io.out := out(7,0)   // error - Reference stateReg is not declared.
//io.out := out        // ok

  val one = 1.U

  val stateReg = RegInit(one)
  when (stateReg === one) {
    stateReg := 2.U
    out := one
  }
}
```
produces:
```
 ...
[info] [0.001] Elaborating design...
[info] [1.052] Done elaborating.
Total FIRRTL Compile Time: 302.6 ms
[error] (run-main-0) firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[RefBeforeDef.scala 16:18:@13.4]: [module RefBeforeDef] Reference stateReg is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[RefBeforeDef.scala 16:18:@13.4]: [module RefBeforeDef] Reference stateReg is not declared.
 ...
```
if `io.out := out(7,0)` is replaced by `io.out := out`, the error goes away

ucbjrl, 2019-03-08:
This was fixed in the master branch with #868 - and has been released with SNAPSHOTS since December, 2018.

closed 2019-03-08 by ucbjrl