chipsalliance/firrtl #1049: Branch condition optimization

donggyukim, 2019-03-09:
**Type of issue**:  other enhancement

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**
  * **What is the current behavior?**
I noticed that FIRRTL generates very ugly verilog code for Hwacha's MasterSequencer:
https://github.com/ucb-bar/hwacha/blob/master/src/main/scala/sequencer-master.scala

```
    if (reset) begin
      v_0 <= 1'h0;
    end else begin
      if (reset) begin // unreachable
        v_0 <= 1'h0;
      end else begin
        if (_T_60800) begin
          if (3'h0 == head) begin
            v_0 <= 1'h0;
          end else begin
            if (_T_4035) begin
              if (io_op_bits_active_vst) begin
                if (3'h0 == _T_3906) begin
                  v_0 <= 1'h1;
                end else begin
                  if (3'h0 == _T_3904) begin
                    v_0 <= 1'h1;
                  end else begin
                    if (3'h0 == tail) begin
                      v_0 <= 1'h1;
                    end else begin
                      if (io_op_bits_active_vld) begin
                        if (3'h0 == _T_3906) begin
                          v_0 <= 1'h1;
                         ... // very long!!!
```
This shows a fraction of the update for ` v_0`, which is much longer and deeper. The second `reset` is duplicate. This update can be much shorter.

I'm also worried about #261, which probably needs to be high priority.

  * **What is the expected behavior?**

We may want to const-prop mux trees like this with the following rules:

* `MUX(a, 1, b) => OR(a, b)`
* `MUX(a, 0, b) => OR(NOT(a), b)`
* `MUX(a, b, 1) => OR(NOT(a), b)`
* `MUX(a, b, 0) => OR(a, b)`

For #261, I believe we should at least support an optimization with patten match.

  * **Please tell us about your environment:**

one of a machines

**What is the use case for changing the behavior?**

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**:  proposal

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

donggyukim, 2019-03-10:
More rules:
* `MUX(a, x, MUX(b, x, y)) => MUX(OR(a, b), x, y)`
* `MUX(a, x, MUX(b, y, x)) => MUX(OR(a, NOT(b)), x, y)`
* `MUX(a, MUX(b, x, y), x) => MUX(OR(NOT(a), b), x, y)`
* `MUX(a, MUX(b, y, x), x) => MUX(OR(NOT(a), NOT(b)), x, y)`

aswaterman, 2019-03-10:
`MUX(a, 1, b) => OR(a, b)` and `MUX(a, b, 0) => AND(a, b)` are probably always worthwhile substitutions.  The other ones, I'm not so sure about.

I'm generally opposed to optimizations that reshape Mux trees, as this removes control from the programmer.  The synthesis tools don't always do the optimal thing, and rewriting code to change how muxes are expressed can sometimes make a significant difference.  This proposed change would obstruct such efforts.

Also keep in mind that some of these proposed changes will introduce more nodes, and so might hurt compile time and simulation runtime, without accomplishing much.  (e.g. `MUX(a, 0, b) => OR(NOT(a), b)`)

donggyukim, 2019-03-10:
Using two rules mentioned by @aswaterman, the above tree becomes prettier:
```
  always @(posedge clock) begin
    if (reset) begin
      v_0 <= 1'h0;
    end else begin
      if (reset) begin
        v_0 <= 1'h0;
      end else begin
        if (_T_60800) begin
          if (3'h0 == head) begin
            v_0 <= 1'h0;
          end else begin
            if (_T_4035) begin
              if (io_op_bits_active_vst) begin
                v_0 <= _GEN_29252;
              end else begin
                if (io_op_bits_active_vld) begin
                  v_0 <= _GEN_27354;
                end else begin
                  if (io_op_bits_active_vstx) begin
                    v_0 <= _GEN_25048;
                  end else begin
                    if (io_op_bits_active_vldx) begin
                      v_0 <= _GEN_22854;
                    end else begin
                      if (io_op_bits_active_vamo) begin
                        v_0 <= _GEN_20396;
                        ...
```

Also it seems that other rules have bad side effects. Implementing the two rules is trivial, so I punt it to @jackkoenig.

albert-magyar, 2020-03-09:
The agreed-upon rules were added in #1052.

closed 2020-03-09 by albert-magyar