chipsalliance/firrtl #1053: SingleTargetAnnotation is constrained to be subtype of deprecated Named

chick, 2019-03-12:
## Type of issue:
Correctness issue

## Description
Named is deprecated.
SingleTargetAnnotation is parmeterized to be [T <: Named].
I believe this should be changed to [T <: Target]
Also the scaladoc for NoTargetAnnotation should have it's reference to Named fixed.

## Impact
No functional change. Just making recent API additions consistent

johnsbrew, 2019-03-12:
Not so sure this has no impact on existing code.
We currently have: `sealed trait Target extends Named`
As far as I understood when dealing with `Target`, `Named` is the legacy `Target` and is still in use in numerous place.
Most notably when annotating a chisel node I had to use the following:

```scala
case class MyAnnotation(target: InstanceId)
    extends ChiselAnnotation {
  def toFirrtl = MyFirAnnotation(target.toNamed)
}

```
 Notice the `target.toNamed`: it would not work with `SingleTargetAnnotation[T <: Target]`.
The point is that all legacy constructs using `Named` should migrate to `Target` including the `InstanceId` used in chisel.

```scala
trait InstanceId {
  def instanceName: String
  def pathName: String
  def parentPathName: String
  def parentModName: String
  /** Returns a FIRRTL Named that refers to this object in the elaborated hardware graph */
  def toNamed: Named
}
 ``` 
I far as I search for `def toNamed` in chisel source it would not be that difficult to upgrade its few but I am afraid that some side pieces of code would get broken along the way...
So although it is highly desirable, I guess the migration will take quite a while.

My conclusion is that it is not (currently) a firrtl issue but a chisel issue (and potentially other front-ends issue?) 
I would label this issue `low priority` and `API Modification`.

ucbjrl, 2019-07-17:
Pr #1122 attempts to implement the conversion from `Named` to `Target` but has brought to light several issues:
- serialization of annotations: we may need to support the old `Named` serialization since clients may have existing annotation files they want to use and existing ScalaTests use this (`Named`) format for annotation resources. This is easy,
- `DedupModules.buildRTLTags()` generates `module2Annotations` by calling `Annotation.getTargets()` which returns `Annotation[Target]`-typed annotations of which there is currently only one: `DontTouchAnnotation` (the rest are `Annotation[Named]`). This logic breaks if every `Annotation` is of type `Annotation[Target]`. This may be harder, depending on the intended use of `module2Annotations`.