chipsalliance/firrtl #1059: Errors when tried to genearte verilog

JaeHun0304, 2019-03-18:
**Type of issue**: bug report
OS: windows x64
version: chisel3
I created my own hardware class for testing and compiled successfully. However, I got following errors when I tried to generate verilog code based on the instructions on [frequently asked questions section.](https://github.com/freechipsproject/chisel3/wiki/Frequently-Asked-Questions#get-me-verilog). Also, I created chisel project based on the template on [here](https://github.com/freechipsproject/chisel-template)

### My code
```
package tree

import chisel3._

class TREE extends Module {
	val io = IO(new Bundle {
		val in = Input(UInt(16.W))
		val out = Output(UInt(16.W))
	})

	val regA = RegInit(0.U(16.W))	//temporary results will be stored in registers
	val regB = RegInit(0.U(16.W))
	val regC = RegInit(0.U(16.W))

	when(io.in < 100.U){regA := io.in}	//classify input based on the generated tree
		.otherwise{regB := io.in}

	when(regA < 80.U){regC := regA}
		.otherwise{io.out := 2.U}

	when(regB < 130.U){io.out := 3.U}
		.otherwise{io.out := 4.U}

	when(regC < 50.U){io.out := 0.U}
		.otherwise{io.out := 1.U}

}

object TREE extends App {
  chisel3.Driver.execute(args, () => new TREE)
}
```

### Error logs:

[info] Packaging C:\Users\jjh44\Desktop\Github\CSCE614__Project\target\scala-2.11\csce614__project_2.11-3.1.1.jar ...
[info] Done packaging.
[info] Running tree.TREE
[info] [0.000] Elaborating design...
[info] [0.096] Done elaborating.
[error] (run-main-2) firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error] firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error]         at firrtl.Utils$.error(Utils.scala:396)
[error]         at firrtl.Utils$.throwInternalError(Utils.scala:178)
[error]         at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:237)
[error]         at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:209)
[error]         at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:138)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error]         at logger.Logger$.makeScope(Logger.scala:136)
[error]         at firrtl.Driver$.execute(Driver.scala:209)
[error]         at chisel3.Driver$.execute(Driver.scala:182)
[error]         at chisel3.Driver$.execute(Driver.scala:202)
[error]         at tree.TREE$.delayedEndpoint$tree$TREE$1(TREE.scala:31)
[error]         at tree.TREE$delayedInit$body.apply(TREE.scala:30)
[error]         at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error]         at scala.App$$anonfun$main$1.apply(App.scala:76)
[error]         at scala.App$$anonfun$main$1.apply(App.scala:76)
[error]         at scala.collection.immutable.List.foreach(List.scala:392)
[error]         at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error]         at scala.App$class.main(App.scala:76)
[error]         at tree.TREE$.main(TREE.scala:30)
[error]         at tree.TREE.main(TREE.scala)
[... 121 lines cut ...]
[error]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error]         at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[error]         at sbt.Run.invokeMain(Run.scala:93)
[error]         at sbt.Run.run0(Run.scala:87)
[error]         at sbt.Run.execute$1(Run.scala:65)
[error]         at sbt.Run.$anonfun$run$4(Run.scala:77)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error]         at sbt.TrapExit$App.run(TrapExit.scala:252)
[error]         at java.base/java.lang.Thread.run(Thread.java:834)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error]         at sbt.Run$.executeTrapExit(Run.scala:124)
[error]         at sbt.Run.run(Run.scala:77)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error]         at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at scala.util.Try$.apply(Try.scala:209)
[error]         at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error]         at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
[error]         at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
[error]         at java.base/java.lang.Thread.run(Thread.java:834)
sbt:CSCE614__Project[error] (Compile / runMain) Nonzero exit code: 1
> [error] Total time: 6 s, completed 2019. 3. 17. ?? 11:20:27

edwardcwang, 2019-03-18:
Can reproduce the error with Chisel 3.1.0 on Binder.

JaeHun0304, 2019-03-18:
I am so sorry but I am a newbie for about the Chisel, so could you explain how can I reproduce the error on Binder?, and I checked "build.sbt" file and the Chisel version that I am using is "3.1.1" with Scala version "2.11.12" right now.

jackkoenig, 2019-03-18:
I believe this is fixed by https://github.com/freechipsproject/firrtl/pull/1039 which is not yet in a published release.

@JaeHun0304 The bug occurs when you incompletely connect to a port or wire (we often call this "initialization" even though it doesn't have anything to do with reset) even if you later fully connect to it.

For example:
```scala
val x = Wire(UInt())
when (en) { x := 123.U }
```
This is an error because `x` only has a connection under the condition `en`. Due to a (now fixed) bug, if you full connect later, you'll get a crazy error like you saw. eg.
```scala
val x = Wire(UInt())
when (en) { x := 123.U }
x := 456.U
```
You can work around this either by using the newest snapshot of Chisel: `3.2-031419-SNAPSHOT`.
You can alternatively work around the bug in your code. The problematic part is:
```scala
	when(regA < 80.U){regC := regA}
		.otherwise{io.out := 2.U}
```
In isolation, that would be an incomplete connection to `io.out`, but later you fully connect so it shouldn't error.

Given that this is such inscrutable error message for otherwise legitimate code, do y'all think we should do a `3.1.7` to include this fix?

jackkoenig, 2019-03-18:
Unrelated to the error but worth noting, in your code you have:
```scala
	when(regB < 130.U){io.out := 3.U}
		.otherwise{io.out := 4.U}

	when(regC < 50.U){io.out := 0.U}
		.otherwise{io.out := 1.U}
```
Due to last connect semantics, the result you'll see for `io.out` is simply:
```scala
io.out := Mux(recC < 50.U, 0.U, 1.U))
```
Note that only the `when` with `regC < 50.U` is represented, the previous `when` is overridden by last connect semantics.

I'm not sure if this is your desired behavior, but I wanted to let you know.

JaeHun0304, 2019-03-20:
Thanks for the detailed explanation about the cause of the error and possible suggestions for refactoring the code. Using "Mux" is a better way and appropriate for my intent. I didn't know that I can put the selector based on comparing the register value. Thanks for your reply. I am going to work around with the Chisel "3.2-031419-SNAPSHOT" version then. Thanks!

JaeHun0304, 2019-03-20:
Could you help me with changing the Chisel version? I guess that I should modify the "build.sbt" file's version :=. But, it doesn't work.

edwardcwang, 2019-03-20:
Did you either `reload` in sbt or restart sbt?

jackkoenig, 2019-03-20:
For the record, you don't need to switch to `Mux`, you should just think about restructuring your code. For example, did you can write something like:
```scala
	when(regB < 130.U){io.out := 3.U}
	.elsewhen(regC < 50.U){io.out := 0.U}
        .otherwise{io.out := 1.U}
```
You probably just need to use `elsewhen` and enumerate the cases that way.

JaeHun0304, 2019-03-20:
Here's the [build.sbt file](https://github.com/JaeHun0304/CSCE614__Project/blob/master/build.sbt) that I am using.
I just tried to reload sbt like below:

> [jaehun@jaehun-pc CSCE614__Project]$ sbt
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/jaehun/github/CSCE614__Project/project
[info] Loading settings from build.sbt ...
[info] Set current project to CSCE614__Project (in build file:/home/jaehun/github/CSCE614__Project/)
[info] sbt server started at local:///home/jaehun/.sbt/1.0/server/1a8703b2d20e4e5f8fa3/sock
sbt:CSCE614__Project> 
sbt:CSCE614__Project> 
sbt:CSCE614__Project> reload
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/jaehun/github/CSCE614__Project/project
[info] Loading settings from build.sbt ...
[info] Set current project to CSCE614__Project (in build file:/home/jaehun/github/CSCE614__Project/)

However, I still have "scala.MatchError: WVoid (of class firrtl.WVoid$)" error right now.

jackkoenig, 2019-03-20:
So `version` is the version of your project, to change the chisel3 version, change:
`"chisel3" -> "3.1.+",` to `"chisel3" -> "3.2-031419-SNAPSHOT",`

JaeHun0304, 2019-03-20:
I changed the default chisel3 version as @jackkoenig suggested and the " Caused by: scala.MatchError: WVoid" error has been disappeared. However, when I tried to run example code of [chisel template](https://github.com/freechipsproject/chisel-template) which is "test:runMain gcd.GCDMain" I got another error that said "NoClassDefFoundError". I guess that it does not properly load the chisel3 cores after changing the version. Is there any other configuration that I have to do other than chaning the build.sbt batch file? Below is the terminal log for that:

> sbt:CSCE614__Project> test:runMain gcd.GCDMain
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running gcd.GCDMain 
[info] [0.002] Elaborating design...
[info] [0.613] Done elaborating.
Total FIRRTL Compile Time: 246.9 ms
[error] (run-main-5) java.lang.NoClassDefFoundError: chisel3/core/ImplicitModule
[error] java.lang.NoClassDefFoundError: chisel3/core/ImplicitModule
[error] 	at chisel3.iotesters.setupTreadleBackend$.apply(TreadleBackend.scala:141)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply$mcZ$sp(Driver.scala:54)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:39)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:39)
[error] 	at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:138)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at logger.Logger$.makeScope(Logger.scala:136)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1.apply$mcZ$sp(Driver.scala:39)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:39)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:39)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at chisel3.iotesters.Driver$.execute(Driver.scala:38)
[error] 	at chisel3.iotesters.Driver$.execute(Driver.scala:100)
[error] 	at gcd.GCDMain$.delayedEndpoint$gcd$GCDMain$1(GCDMain.scala:27)
[error] 	at gcd.GCDMain$delayedInit$body.apply(GCDMain.scala:26)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.collection.immutable.List.foreach(List.scala:392)
[error] 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error] 	at scala.App$class.main(App.scala:76)
[error] 	at gcd.GCDMain$.main(GCDMain.scala:26)
[error] 	at gcd.GCDMain.main(GCDMain.scala)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.lang.reflect.Method.invoke(Method.java:498)
[error] 	at sbt.Run.invokeMain(Run.scala:93)
[error] 	at sbt.Run.run0(Run.scala:87)
[error] 	at sbt.Run.execute$1(Run.scala:65)
[error] 	at sbt.Run.$anonfun$run$4(Run.scala:77)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] 	at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] Caused by: java.lang.ClassNotFoundException: chisel3.core.ImplicitModule
[error] 	at java.net.URLClassLoader.findClass(URLClassLoader.java:382)
[error] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
[error] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
[error] 	at chisel3.iotesters.setupTreadleBackend$.apply(TreadleBackend.scala:141)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply$mcZ$sp(Driver.scala:54)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:39)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:39)
[error] 	at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:138)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at logger.Logger$.makeScope(Logger.scala:136)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1.apply$mcZ$sp(Driver.scala:39)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:39)
[error] 	at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:39)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at chisel3.iotesters.Driver$.execute(Driver.scala:38)
[error] 	at chisel3.iotesters.Driver$.execute(Driver.scala:100)
[error] 	at gcd.GCDMain$.delayedEndpoint$gcd$GCDMain$1(GCDMain.scala:27)
[error] 	at gcd.GCDMain$delayedInit$body.apply(GCDMain.scala:26)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.collection.immutable.List.foreach(List.scala:392)
[... 5 lines cut ...]
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[... cut ...]
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.Try$.apply(Try.scala:209)
[error] 	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (Test / runMain) Nonzero exit code: 1
[error] Total time: 1 s, completed Mar 20, 2019 6:00:25 AM

edwardcwang, 2019-03-20:
Did you also bump the chisel-testers version?

JaeHun0304, 2019-03-20:
Oh, I did not. The current chisel-testers version is "[1.2.5,1.3-SNAPSHOT[" It seems has different version numbering with chisel3 version. To what version should I change the testers?

edwardcwang, 2019-03-20:
Try `1.3-SNAPSHOT` and let us know if it works.

JaeHun0304, 2019-03-20:
It works right now. Really thanks for your help. Test seed and cycle passed successfully.

albert-magyar, 2019-11-05:
Closing as resolved

closed 2019-11-05 by albert-magyar