chipsalliance/firrtl #1070: Missing LICENSE file

johnsbrew, 2019-04-10:
**Type of issue**:  documentation

**Impact**: no functional change

**Development Phase**: request

**Other information**

**What is the current behavior?** ? No LICENSE file at root of repository despite mentioned in every single file within the repository

**What is the expected behavior?** LICENSE file (or multiple ones as for chisel3 repository)

chick, 2019-04-10:
The license file is in `src/`. I believe this was modeled after some other project though I cannot remember which one. Anyone else remember? @jackkoenig?

jackkoenig, 2019-04-10:
I believe firrtl was modeled on chisel, but I have no idea why it was that way. We should move it to the root, and we need to add the Apache 2.0 license which is what future contributions by non-University of California members are contributed under.

johnsbrew, 2019-04-10:
Sorry about the misleading comparison with chisel where the LICENSE.txt is also in `./src`.

I actually intended to refer to riscv-boom repo where there are 2 files at repository root with LICENSE (berkeley) and LICENSE.SiFive (Apache 2.0)