chipsalliance/firrtl #1077: RemoveWires not respecting SSA for initializations

seldridge, 2019-04-26:
**Type of issue**: bug report

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Compile the following:
```
circuit Foo :
  module Foo :
    input clock : Clock
    input foo : UInt<1>
    reg y : UInt<1>, clock with :
      reset => (UInt<1>("h0"), y_fault)
    node y_fault = foo
```
You get:

```
Exception in thread "main" firrtl.passes.CheckChirrtl$UndeclaredReferenceException: : [module Foo] Reference y_fault is not declared.
```

This is a thing I didn't fix in https://github.com/freechipsproject/firrtl/pull/823. "There should still be a problem with reset dependencies." :neutral_face: 

  * **What is the current behavior?**

Initialization definition placed after use (SSA violation).

  * **What is the expected behavior?**

The `node y_fault = foo` should be defined before use.

  * **Please tell us about your environment:**
    - version: FIRRTL v1.1.7 or master

**What is the use case for changing the behavior?**

This should work. This isn't the pathological case of a loop that you can't do something about, e.g., https://github.com/freechipsproject/firrtl/issues/749#issuecomment-369820422.

**Impact**: no functional change

**Development Phase**: request

**Other information** 

Affecting:
  - https://github.com/IBM/chiffre/issues/30

jackkoenig, 2019-04-26:
I'm a little confused, the FIRRTL you gave is illegal FIRRTL as is and errors long before RemoveWires. Are you saying that RemoveWires can create such cases?

seldridge, 2019-04-26:
Good catch. You're right. It is illegal FIRRTL.

This is my fault injection stuff that's generating it. I thought that blind right hand side reference replacement was safe, but that is (for some reason) running after `RemoveReset`. At that point, the initialization isn't the real initialization. Investigating. This is likely my error.

seldridge, 2019-04-26:
Alright, `RemoveWires` can actually create such a case... 
```scala
  it should "not result in invalid FIRRTL" in {
    val result = compileBody(
      s"""|input clock : Clock
          |input foo : UInt<2>
          |output bar : UInt<2>
          |wire y_fault : UInt<2>
          |reg y : UInt<2>, clock with :
          |  reset => (UInt<1>("h0"), y_fault)
          |y_fault <= foo
          |bar <= y
          |""".stripMargin)
    passes.CheckHighForm.execute(result)
  }
```

The output of `RemoveWires` is then:

```
circuit Foo : 
  module Foo : 
    input clock : Clock 
    input foo : UInt<2> 
    output bar : UInt<2> 
  
    reg y : UInt<2>, clock with :
      reset => (UInt<1>("h0"), y_fault) 
    node y_fault = foo 
    bar <= y 
    y <= y
```

jackkoenig, 2019-04-26:
Welp whoops.

jackkoenig, 2019-05-07:
Fixed by https://github.com/freechipsproject/firrtl/pull/1078

PR #1078 (2019-05-04): Add register init to RemoveWires dependencies
Adds a dependency to `RemoveWires` for registers that do not initialize themselves. (Oddly, adding a dependency for self-initializing registers causes `InfoSpec` to fail.)

Fixes #1077
closed 2019-05-07 by jackkoenig