chipsalliance/firrtl #1084: Support ScalaJS and ScalaNative

jackkoenig, 2019-05-17:
Unlike Chisel, I don't think there's anything in FIRRTL that depends on the JVM, it would be nice if we could support alternative backends for smoother integration with web browsers and native code.

**Type of issue**: feature request

**What is the use case for changing the behavior?**

More support

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

grebe, 2019-05-17:
I started this work, check out my branch [here](https://github.com/grebe/firrtl/tree/scalajs). It's from 12/2018, so not super stale.

colin4124, 2019-05-18:
Now scala native don't support scala 2.12 yet,  but in next few data, scala native would be 0.4.0-M1( support scala 2.12) https://github.com/scala-native/scala-native/milestones

XVilka, 2019-05-27:
Note that Scala Native 0.4.0-M2 was released: https://github.com/scala-native/scala-native/releases/tag/v0.4.0-M2