chipsalliance/firrtl #1097: Support WIR in CHIRRTL Transforms

jackkoenig, 2019-06-17:
We can fix this with simply a ToIR transform that runs at the beginning of the CHIRRTL transforms

**Type of issue**: bug report

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**
  * **What is the current behavior?**
  * **What is the expected behavior?**
  * **Please tell us about your environment:**
    (examples)
    - version: `3.0-SNAPSHOT`
    - OS: `Linux knight 4.4.0-92-generic #115-Ubuntu SMP Thu Aug 10 09:04:33 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`
    - verilator version (if relevant)

**What is the use case for changing the behavior?**

**Impact**: no functional change

**Development Phase**: request

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

jackkoenig, 2019-07-01:
Worked around in https://github.com/freechipsproject/treadle/pull/106

albert-magyar, 2020-03-02:
Subsumed by #1350 and #1410

PR #1350 (2020-05-26): Remove WIR nodes that "shadow" standard IR nodes
**Type of improvement:** API simplification + pass reusability improvement
**API impact:** API modification (see below)
**Backend Code Generation Impact:** none
**Desired merge strategy:** rebase
**Release notes:**
_See extended commit message of ba40a5c2cdbe5e485bbcefe1411ab551b78c7cfa for lengthy release notes._

Fixes #1097 
Helps with #559 
Removes incompatibility of WIR with ProtoBuf from #832

This PR absorbs the extra fields of `WRef`, `WSubField`, `WSubIndex`, `WSubAccess`, and `WDefInstance` into the corresponding standard IR nodes. The [... cut]
closed 2020-03-02 by albert-magyar