chipsalliance/firrtl #1161: CheckHighFormLike: Reference memory is not declared

ekiwi, 2019-08-20:
Somehow using nodes to address `smem` breaks in  `firrtl.passes.CheckHighFormLike`.

Minimized Chisel3:
```.scala
class regfile extends Module {
  val io = IO(new Bundle{
    val sel = Input(Bool())
    val addr = Input(UInt(5.W))
  })

  val raddr = Mux(io.sel, io.addr, 0.U)
  val memory = SyncReadMem(32, UInt(2.W))
  val rdata = memory.read(raddr)
}
```

Chirrtl:
```
circuit regfile :
  module regfile :
    input clock : Clock
    input reset : UInt<1>
    output io : { flip sel : UInt<1>, flip addr : UInt<5>}
  
    node raddr = mux(io.sel, io.addr, UInt<1>("h0")) @[regfile.scala 30:18]
    smem memory : UInt<2> [32] @[regfile.scala 31:27]
    read mport rdata = memory[raddr], clock @[regfile.scala 32:26]
```

Compiler error + trace:
```
 @[regfile.scala 30:18]: [module regfile] Reference memory is not declared.
firrtl.passes.CheckHighFormLike$UndeclaredReferenceException:  @[regfile.scala 30:18]: [module regfile] Reference memory is not declared.
	at firrtl.passes.CheckHighFormLike.checkHighFormE$1(Checks.scala:158)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$16(Checks.scala:169)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$16$adapted(Checks.scala:169)
	at firrtl.WSubField.foreachExpr(WIR.scala:59)
	at firrtl.traversals.Foreachers$ExprForMagnet$$anon$6.foreach(Foreachers.scala:44)
	at firrtl.traversals.Foreachers$ExprForeach$.foreach$extension(Foreachers.scala:54)
	at firrtl.passes.CheckHighFormLike.checkHighFormE$1(Checks.scala:169)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$16(Checks.scala:169)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$16$adapted(Checks.scala:169)
	at firrtl.WSubField.foreachExpr(WIR.scala:59)
	at firrtl.traversals.Foreachers$ExprForMagnet$$anon$6.foreach(Foreachers.scala:44)
	at firrtl.traversals.Foreachers$ExprForeach$.foreach$extension(Foreachers.scala:54)
	at firrtl.passes.CheckHighFormLike.checkHighFormE$1(Checks.scala:169)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$23(Checks.scala:210)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$23$adapted(Checks.scala:210)
	at firrtl.ir.Connect.foreachExpr(IR.scala:390)
	at firrtl.traversals.Foreachers$StmtForMagnet$$anon$2.foreach(Foreachers.scala:21)
	at firrtl.traversals.Foreachers$StmtForeach$.foreach$extension(Foreachers.scala:35)
	at firrtl.passes.CheckHighFormLike.checkHighFormS$1(Checks.scala:210)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$25(Checks.scala:211)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$25$adapted(Checks.scala:211)
	at scala.collection.immutable.List.foreach(List.scala:388)
	at firrtl.ir.Block.foreachStmt(IR.scala:363)
	at firrtl.traversals.Foreachers$StmtForMagnet$$anon$1.foreach(Foreachers.scala:18)
	at firrtl.traversals.Foreachers$StmtForeach$.foreach$extension(Foreachers.scala:35)
	at firrtl.passes.CheckHighFormLike.checkHighFormS$1(Checks.scala:211)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$25(Checks.scala:211)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$25$adapted(Checks.scala:211)
	at scala.collection.Iterator.foreach(Iterator.scala:937)
	at scala.collection.Iterator.foreach$(Iterator.scala:937)
	at scala.collection.AbstractIterator.foreach(Iterator.scala:1425)
	at scala.collection.IterableLike.foreach(IterableLike.scala:70)
	at scala.collection.IterableLike.foreach$(IterableLike.scala:69)
	at scala.collection.AbstractIterable.foreach(Iterable.scala:54)
	at firrtl.ir.Block.foreachStmt(IR.scala:363)
	at firrtl.traversals.Foreachers$StmtForMagnet$$anon$1.foreach(Foreachers.scala:18)
	at firrtl.traversals.Foreachers$StmtForeach$.foreach$extension(Foreachers.scala:35)
	at firrtl.passes.CheckHighFormLike.checkHighFormS$1(Checks.scala:211)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$37(Checks.scala:236)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$37$adapted(Checks.scala:236)
	at firrtl.ir.Module.foreachStmt(IR.scala:689)
	at firrtl.traversals.Foreachers$ModuleForMagnet$$anon$11.foreach(Foreachers.scala:79)
	at firrtl.traversals.Foreachers$ModuleForeach$.foreach$extension(Foreachers.scala:92)
	at firrtl.passes.CheckHighFormLike.checkHighFormM$1(Checks.scala:236)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$41(Checks.scala:246)
	at firrtl.passes.CheckHighFormLike.$anonfun$run$41$adapted(Checks.scala:246)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:58)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:51)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:47)
	at firrtl.passes.CheckHighFormLike.run(Checks.scala:246)
	at firrtl.passes.CheckHighFormLike.run$(Checks.scala:69)
	at firrtl.passes.CheckHighForm$.run(Checks.scala:259)
	at firrtl.passes.Pass.execute(Passes.scala:23)
	at firrtl.passes.Pass.execute$(Passes.scala:21)
	at firrtl.passes.CheckHighForm$.execute(Checks.scala:259)
	at firrtl.Transform.$anonfun$runTransform$2(Compiler.scala:216)
	at firrtl.Utils$.time(Utils.scala:183)
	at firrtl.Transform.runTransform(Compiler.scala:216)
	at firrtl.SeqTransformBased.$anonfun$runTransforms$1(Compiler.scala:281)
	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:122)
	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:118)
	at scala.collection.immutable.List.foldLeft(List.scala:85)
	at firrtl.SeqTransformBased.runTransforms(Compiler.scala:281)
	at firrtl.SeqTransformBased.runTransforms$(Compiler.scala:280)
	at firrtl.SeqTransform.runTransforms(Compiler.scala:285)
	at firrtl.SeqTransform.execute(Compiler.scala:291)
	at firrtl.Transform.$anonfun$runTransform$2(Compiler.scala:216)
	at firrtl.Utils$.time(Utils.scala:183)
	at firrtl.Transform.runTransform(Compiler.scala:216)
```

ekiwi, 2019-08-20:
Confirmation that the fact that `raddr` is a `node` and not a `wire` is the problem: Adding a `WireDefault` makes the circuit compile.

The following works:
```.scala
class regfile extends Module {
  val io = IO(new Bundle{
    val sel = Input(Bool())
    val addr = Input(UInt(5.W))
  })

  val raddr = WireDefault(Mux(io.sel, io.addr, 0.U))
  val memory = SyncReadMem(32, UInt(2.W))
  val rdata = memory.read(raddr)
}
```

albert-magyar, 2019-10-29:
Closed and absorbed into #727

closed 2019-10-29 by albert-magyar