chipsalliance/firrtl #1171: PadWidths: generates illegal widths for Dshl

ekiwi, 2019-09-04:
This line in `PadWidths` ensures that the argument `e` width of a `dshl` is the same as its result width:
https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/PadWidths.scala#L46
```.scala
    case Dshl =>
        // special case as args aren't all same width
        ex copy (op = Dshlw, args = Seq(fixup(width(ex.tpe))(ex.args.head), ex.args(1)))
```

(i.e. `Result Width == w_e` instead of `Result Width == w_e + 2^(w_n) - 1`)

This is incorrect according to the firrtl spec:
[image]

I would suggest just removing this case altogether.
However, I am not sure why it was added in the first place.

@donggyukim and @jackkoenig, maybe you can comment on this.

aswaterman, 2019-09-04:
Are you sure?  It's taking the width of `ex`, which should already include the `+2^n-1`, since this runs after InferWidths.

ekiwi, 2019-09-04:
> Are you sure? It's taking the width of `ex`, which should already include the `+2^n-1`, since this runs after InferWidths.

The problem is that after we extend `e` to be the same width as the result (i.e., `e_w == Result Width`) the equation given in the spec (`Result Width == w_e + 2^(w_n) - 1`) is no longer true.

In other words, once we force `w_e` to be equivalent to the `Result Width`,  `w_n` must be `1` in order for the equation from the spec to hold true.
Of cause `w_n` is not always `1`, thus we get an ill-typed expression for any `n` that is wider than `1`.

aswaterman, 2019-09-04:
Well, we need this for Verilog emission (its rules for `<<` don't agree with Firrtl's for `dshl`).  So we either need to say that PadWidths is part of Verilog emission (so that it's allowed to generate invalid nodes) or reorganize things accordingly.

ekiwi, 2019-09-04:
> Well, we need this for Verilog emission (its rules for `<<` don't agree with Firrtl's for `dshl`). So we either need to say that PadWidths is part of Verilog emission (so that it's allowed to generate invalid nodes) or reorganize things accordingly.

That seems like a good idea. I stumbled over this bug because I assumed `LowFirrtlOptimization` would produce valid `firrtl`.

ekiwi, 2019-09-04:
As we add more backends for firrtl such as `treadle`, the UCLID5 model checking backend from @albert-magyar and my symbolic execution engine, it would be great to separate the Verilog and the LowFirrtl emission more clearly.

aswaterman, 2019-09-04:
Yeah.  The alternative that comes to mind is to introduce new IR nodes that match Verilog semantics, so that the IR remains legal through the final passes.  At first blush, that seems too heavyweight if we can continue to contain this stuff within the Verilog backend.  (But there might be some additional utility to that approach that I'm not seeing.)

seldridge, 2019-09-04:
For all intents and purposes `PadWidths` is currently a part of Verilog emission. If you run with `-X low` the low emitter runs after `MidFormToLowForm`. If you run with `-X verilog` then `PadWidths` and other optimizations run. The resulting FIRRTL circuit state will include those optimizations. That optimized low circuit is then fed into the emitter where additional optimizations happen (`VerilogPrep`, etc.), but those optimizations are *not* included in the circuit state.

[Fundamentally, the codebase is ambiguous on what low form means.](https://github.com/freechipsproject/firrtl/issues/1165) My knee-jerk reaction is that low form means "after `MidFormToLowForm`" and optimizations should be rolled into the Verilog emitter. There are complications with this approach as there exist transforms that are "LowForm", but are really supposed to run after pre-emitter optimizations.

ekiwi, 2019-09-04:
> For all intents and purposes `PadWidths` is currently a part of Verilog emission

Treadle currently uses these passes (https://github.com/freechipsproject/treadle/blob/master/src/main/scala/treadle/stage/phases/PrepareAst.scala#L39):
```.scala
Seq(new LowFirrtlOptimization, new BlackBoxSourceHelper, new FixupOps, AugmentPrintf)
```

`PadWidths` is run as part of the `LowFirrtlOptimization`.
Judging from the name of `LowFirrtlOptimization` I expected this to lower to (legal) `LowFirrtl` and also run some optimizations. Why does this run transformations that are part of the `Verilog` emitter?

What should `treadle` use to get simplified + optimized but still legal `LoFIRRTL`?

## Wrt. the "meaning" of `LowForm`:

`LoFIRRTL` is defined by the spec: https://github.com/freechipsproject/firrtl/blob/8c83631b1e7296b03420d807144c21bc89df79d2/spec/spec.tex#L1616

If there are additional restrictions on `LoFIRRTL` they should be updated.

Transformations that violate `FIRRTL` semantics should be kept separate from others. Maybe in a special `firrtl.backends.verilog` package. It can be very confusing when the FIRRTL emitted suddenly doesn't make sense in terms of the spec anymore.

seldridge, 2019-09-04:
Ah, fair point. I forgot about Treadle usage here.

I think you've identified that `LowFirrtlOptimizations` are not producing `LowForm` but are something that both Treadle and the `VerilogEmitter` require. Note that `LowFirrtlOptimizations` are additionally *more* restrictive in other ways (all `IsInvalid`s are removed), but additional restrictions would still be a `LowForm` subset. 

More concretely, I think you're divining the true dependencies here. It looks like `PadWidths` invalidates `LowForm`. `Treadle` and the `VerilogEmitter` require `PadWidths`, but they *don't* require `LowForm`, just a set of optimization transforms that require `LowForm`. Even so, `LowFirrtlOptimizations` are listed as outputting `LowForm` so any custom transforms that are `inputForm=LowForm` will run after it and may rely on things like what `PadWidths` is doing. :woman_shrugging: The dependency API lets you specify this explicitly, but it's ugly to try and sort out what the actual dependencies are...

ekiwi, 2019-09-04:
>  It looks like `PadWidths` invalidates `LowForm`

That's the problem. If something gives you a subset of `LoFIRRTL`, that's ok, but if the result is ill typed, then it isn't valid FIRRTL.

jackkoenig, 2019-09-04:
IMO we just need to move the dshlw injection into the Verilog emitter

jackkoenig, 2019-09-04:
In fact, this has come up before but we just never fixed it 😬 https://github.com/freechipsproject/firrtl/issues/498#issuecomment-315474606 and https://github.com/freechipsproject/firrtl/issues/866

ekiwi, 2019-09-04:
> IMO we just need to move the dshlw injection into the Verilog emitter

Thanks! I have to admit I didn't even realize that the op was replaced with `Dshlw`.

ekiwi, 2019-09-04:
If we move `Dshlw` injection into the emitter, treadle could get rid of their `FixupOps` pass: https://github.com/freechipsproject/treadle/blob/master/src/main/scala/treadle/utils/ToLoFirrtl.scala#L36

ekiwi, 2019-09-04:
After looking a bit more at `PadWidths`, there are more operations where the result might be an ill-typed expression: `Mul`, `Div`, and `Rem`

The pass also looks at some unary operations which doesn't make sense to me. Am I overlooking a case in which the `fixup` would change the argument to a `Not` or `Shr` operation?

seldridge, 2019-10-22:
@ekiwi: A result of this asymmetry is that `InferTypes` (which relies on specification-defined width propagation) is not usable after `PadWidths`.

Currently, this is biting me for `Rem`: `VerilogMemDelays` (seemingly after https://github.com/freechipsproject/firrtl/pull/1202) now produces `UnknownType` and `UnknownFlow`. With the Dependency API 2 PR, `Legalize` needs to run after `VerilogMemDelays`, but it can't handle `UnknownType`. I want to run `InferTypes` to clean this up, but the types have been mangled by `PadWidths` to be improper for `Rem`.

Concretely, the following Verilog emitter test fails if you run `InferTypes` after `PadWidths`:

```
circuit Test :
  module Test :
    input in : UInt<8>
    output out : UInt<1>
    out <= rem(in, UInt<1>("h1"))
```

Expected:
```verilog
module Test(
  input  [7:0] in, 
  output  out 
);
  wire [7:0] _GEN_0;
  assign out = _GEN_0[0];
  assign _GEN_0 = in % 8'h1;
endmodule
```

With `InferTypes`:
```
module Test( 
  input [7:0] in,
  output out 
);
  wire [7:0] _GEN_0;
  assign _GEN_0 = in % 8'h1;
  assign out = _GEN_0[7:0];
endmodule
```

The problem is that `PadWidths` is increasing the width of the remainder primop to be the maximum of the two arguments when it's defined in the specification as being the minimum of the two.

More generally , any `inputForm=LowForm` custom transform may not work because of the incorrect WIR here.

albert-magyar, 2020-03-02:
`PadWidths` doesn't actually generate illegal widths, so it doesn't break the spec. It generates "private" IR nodes that may be absorbed into the general FIRRTL spec at a later date.

Subsumed by #1410.

ekiwi, 2020-03-02:
This issue isn't resolved.

The `LowFirrtlOptimization` still claims to emit `LowForm` but uses `PadWidths`: https://github.com/freechipsproject/firrtl/blob/6c7a15a2d9123fd94770dda15f9e9070ae6b2bdc/src/main/scala/firrtl/LoweringCompilers.scala#L114

`PadWidths` should be moved into the Verilog Emitter or into a `LowFirrtlToVerilog` transform.

albert-magyar, 2020-03-02:
The original title of the issue is incorrect, though. I'm not claiming this is resolved, just that the actual underlying issue is being tracked.

Once the title is resolved, it is just a "what should the fate of `Dshlw` be" issue (see #270, #866). This is being tracked in #1410 for inclusion in 1.4.0, the goal of which is to get rid of special internal-only WIR nodes by either incorporating them into the spec of eliminating them work. I am likely to absorb #1410 and #1350 into a single issue to further streamline it.

The issue with `outputForm = LowForm` in `LowFirrtlOptimization` is largely an artifact of the now-deprecated form-based pass-ordering API being fundamentally broken. Schuyler has already fixed a ton of pre-existing bugs in getting #1123 ready, but I think the current idea is to avoid tacking more on for the 1.3.0 release, especially if that work would have to be undone if `Dshlw` is promoted to a first-class IR node.

The current dependency of issues is merge #1123 -> release 1.3.0 -> fix all WIR-related problems in one go.