chipsalliance/firrtl #1175: old SNAPSHOT causes compile error in chisel3 compile

Soya-Onishi, 2019-09-08:
**Type of issue**: bug report(?)

As I mentioned in title, sonatype snapshot does not have `StageError`.
`StageError` is added in Aug 1, but snapshot is published in Aug 13. This seems to causes this problem.

I do not know about library publishing process, and this issue may not be suitable as issue, so this may be impertinence, but If there is no problem, please publish a snapshot to sonatype.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **What is the current behavior?**
compile error in chisel3 due to no existence of `StageError`
  * **What is the expected behavior?**
compile success
  * **Please tell us about your environment:**
    - commit hash in chisel3: `67e90c12`
    - OS: `Linux soya 5.0.0-27-generic #28~18.04.1-Ubuntu SMP Thu Aug 22 03:00:32 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
`

**Impact**: no functional change

seldridge, 2019-09-08:
If you want to compile Chisel `master`, you may need to publish necessary dependencies (`firrtl`) locally.

The snapshots are all expected to work together, but they may not work with `master`.

Can you try removing your local `~/.ivy2/{cache,{local}/edu.berkeley.cs/` directories, cloning FIRRTL `master`, running `sbt publishLocal` in FIRRTL, and then recompiling master Chisel?

Soya-Onishi, 2019-09-09:
> Can you try removing your local ~/.ivy2/{cache,{local}/edu.berkeley.cs/ directories, cloning FIRRTL master, running sbt publishLocal in FIRRTL, and then recompiling master Chisel?

Actually, I've done already. However, I doubted that there is better way before I post this issue.

Thanks for your advice.

closed 2019-09-09 by Soya-Onishi