chipsalliance/firrtl #1178: Internal Error when combinational loop

mariodruiz, 2019-09-13:
**Type of issue**:  feature request

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **What is the current behavior?**
I was doing the Mux4 example, and accidentally connect the output of one multiplexer to its input, and I got an Internal Error.
  * **What is the expected behavior?**
I think I would be better if the compiler returns a warning or error with a combinational loop message. 
  * **Please tell us about your environment:**
    (examples)
    - version: `3.1.0`
    - OS: `Linux ubuntu 4.4.0-31-generic #50-Ubuntu SMP Wed Jul 13 00:07:12 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

**What is the use case for changing the behavior?**

**Impact**: unknown

**Development Phase**: request

sequencer, 2019-09-13:
Have a try on adding `DontCheckCombLoopsAnnotation`.

albert-magyar, 2019-10-24:
Without any additional context to narrow it down to an existing pass, I'm closing this since it will be fixed by #1212

closed 2019-10-24 by albert-magyar