chipsalliance/firrtl #1203: Improve error message for combinational loops

oharboe, 2019-10-16:
The error message below leaves something to be desired when it comes to chasing down a combinational loop.

The trick that I've found is to bring up the generated .fir file and from there, I search for e.g. _T_659 and there will be an annotation in the .fir file that tells me where in the .scala file to look: Thingamabob.scala line 177

Obviosuly it would improve the error message if the references back to Chisel code lines was directly in the error message.

```
node _T_659 = eq(_T_589, UInt<1>("h00")) @[Thingamabob.scala 177:69]
```

```
[deleted]
[info]   firrtl.transforms.CheckCombLoops$CombLoopException: : [module Thingamabob] Combinational loop detected:
[info] Thingamabob._T_589
[info] Thingamabob._T_659
[info] Thingamabob._T_660
[info] Thingamabob._GEN_13
[info] Thingamabob._GEN_43
[deleted]
```

albert-magyar, 2019-10-21:
That's a good idea, I will add the info printing to the error.

oharboe, 2019-10-21:
@albert-magyar Thanks!

commit 991975a8 (2019-10-23): Enhance CheckCombLoops errors with connection info

* Closes #1203
commit 6061fe9e (2019-10-24): Enhance CheckCombLoops errors with connection info

* Closes #1203
commit 0a9a3241 (2019-10-24): Enhance CheckCombLoops errors with connection info

* Closes #1203
closed 2019-10-24 by albert-magyar