chipsalliance/firrtl #1207: Type-parameterized annotations are not deserializable

jwright6323, 2019-10-23:
**Type of issue**: bug report

**What is the use case for changing the behavior?**
If a user tries to build a type-parameterized annotation, chisel will emit it without complaint, but it will not be deserializable by firrtl.

**Impact**: API modification

**Development Phase**: request

**Other information**

For example:
```
trait MyCoolTrait {
  val foo: Int
}

case class MyConcreteClassA(foo: Int) extends MyCoolTrait
case class MyConcreteClassB(foo: Int, bar: String) extends MyCoolTrait

case class MyAnnotation[T <: MyCoolTrait](target: ModuleTarget, data: T)
  extends SingleTargetAnnotation[ModuleTarget] { /* snipped boilerplate */ }
```

This is serialized by chisel without issue, but cannot be deserialized by firrtl because no type hints for `data` are provided. I think there are basically two options:

1. Figure out a smart way to enable this
2. Explicitly disallow it somehow so that chisel errors out when building such a thing

But... I currently have no suggestions on how to do either of these. :)

seldridge, 2019-10-23:
You may want to investigate a custom serializer/deserializer. You get magic see/Des for case classes through json4s but anything more complicated will start to break down. I have not hit the type parametric case before, though! I was aware that function parameters currently don't work either.

jwright6323, 2019-10-23:
Currently I have a custom json4s serializer/deserializer for my case classes, and my annotation takes the serialized string as a parameter, but it seems a bit clunky. I wasn't able to figure out how to get the annotation serializer to correctly (de-)serialize my case classes without modifying firrtl itself.

seldridge, 2019-10-23:
Your use case is custom transforms with custom annotations that need to define how they should be serialized/deserialized without modifying FIRRTL? 

I think we could do this with the `RegisteredTransform` approach where additional serializers/deserializers are defined in the transform and FIRRTL then adds registered serializers/deserializers.

I'll provide a link to the function serialization stuff (https://github.com/freechipsproject/firrtl/issues/847) as these are somewhat related.

jwright6323, 2019-10-23:
> Your use case is custom transforms with custom annotations that need to define how they should be serialized/deserialized without modifying FIRRTL?

That's a good way of describing it, but perhaps we could relax it a bit and say that I just want to specify how some of the fields in the custom annotation are serialized/deserialized rather than the entire annotation.

davidbiancolin, 2019-10-24:
I modified the JsonProtocol (firrtl) to let annotations supply additional TypeHints to get this working in Golden Gate. The fully qualified classnames for everything you need the extra type hint for is captured in an additional annotation, so that deserialization can proceed. I think it would be trivial to let annotations also supply a custom serdes, though i suspect in your case that extra type hints are all you need.

I was going to open a draft PR later today.

albert-magyar, 2019-11-16:
Closed by #1211

closed 2019-11-16 by albert-magyar