chipsalliance/firrtl #1273: line 349:22 mismatched input '(' expecting DEDENT

aabdelsa, 2019-12-06:
i get this error when i compile a .fir file
i don't understand what is the error. i get this file from transforming a scala file into a firrtl. 

circuit Tile :
  module CSR :
    input clock : Clock
    input reset : UInt<1>
    input io_stall : UInt<1>
    input io_cmd : UInt<3>
    input io_in : UInt<32>
    output io_out : UInt<32>
    input io_pc : UInt<32>
    input io_addr : UInt<32>
    input io_inst : UInt<32>
    input io_illegal : UInt<1>
    input io_st_type : UInt<2>
    input io_ld_type : UInt<3>
    input io_pc_check : UInt<1>
    input io_host_fromhost_valid : UInt<1>
    input io_host_fromhost_bits : UInt<32>
    output io_expt : UInt<1>
    output io_evec : UInt<32>
    output io_epc : UInt<32>
    output io_host_tohost : UInt<32>
  
    reg time : UInt<32>, clock with :
      reset => (UInt<1>("h0"), time) @[CSR.scala 105:25]
    reg timeh : UInt<32>, clock with :
      reset => (UInt<1>("h0"), timeh) @[CSR.scala 106:25]
    reg cycle : UInt<32>, clock with :
      reset => (UInt<1>("h0"), cycle) @[CSR.scala 107:25]
    reg cycleh : UInt<32>, clock with :
      reset => (UInt<1>("h0"), cycleh) @[CSR.scala 108:25]
    reg instret : UInt<32>, clock with :
      reset => (UInt<1>("h0"), instret) @[CSR.scala 109:25]
    reg instreth : UInt<32>, clock with :
      reset => (UInt<1>("h0"), instreth) @[CSR.scala 110:25]
    reg PRV : UInt<2>, clock with :
      reset => (UInt<1>("h0"), PRV) @[CSR.scala 119:21]
    reg PRV1 : UInt<2>, clock with :

jackkoenig, 2019-12-06:
This is due to https://github.com/freechipsproject/firrtl/issues/1171, @seldridge, @azidar do we have a plan to fix the `dshlw` issue?

seldridge, 2019-12-06:
I think the plan is to move `VerilogMemDelays` and `PadWidths` out of optimizations and fully into the `VerilogEmitter`. I've superficially tried to do this simply with something like the diff below, but simply moving things runs into `firrtl.EmitterException: Can't emit firrtl.ir.DoPrim as PrimOp argument`. 

```diff
diff --git a/src/main/scala/firrtl/Emitter.scala b/src/main/scala/firrtl/Emitter.scala
index 32027f6..2b0aedd 100644
--- a/src/main/scala/firrtl/Emitter.scala
+++ b/src/main/scala/firrtl/Emitter.scala
@@ -959,6 +959,11 @@ class VerilogEmitter extends SeqTransform with Emitter {
 
   /** Preamble for every emitted Verilog file */
   def transforms = Seq(
+    passes.PadWidths,
+    new firrtl.transforms.ConstantPropagation,
+    passes.Legalize,
+    passes.memlib.VerilogMemDelays, // TODO move to Verilog emitter
+    new firrtl.transforms.ConstantPropagation,
     new BlackBoxSourceHelper,
     new ReplaceTruncatingArithmetic,
     new FlattenRegUpdate,
diff --git a/src/main/scala/firrtl/LoweringCompilers.scala b/src/main/scala/firrtl/LoweringCompilers.scala
index 0d9b971..db9c43c 100644
--- a/src/main/scala/firrtl/LoweringCompilers.scala
+++ b/src/main/scala/firrtl/LoweringCompilers.scala
@@ -110,11 +110,6 @@ class LowFirrtlOptimization extends CoreTransform {
   def transforms = Seq(
     passes.RemoveValidIf,
     new firrtl.transforms.ConstantPropagation,
-    passes.PadWidths,
-    new firrtl.transforms.ConstantPropagation,
-    passes.Legalize,
-    passes.memlib.VerilogMemDelays, // TODO move to Verilog emitter
-    new firrtl.transforms.ConstantPropagation,
     passes.SplitExpressions,
     new firrtl.transforms.CombineCats,
     passes.CommonSubexpressionElimination,
```

One non-solution that might work is to move the `LowFirrtlEmitter` before optimizations. Currently this is getting scheduled like a transform and running "as late as possible" so it'll run with after optimizations if you include them.

My understanding is that the example failure is low FIRRTL that has run through optimizations and then can't be run back through the compiler?

albert-magyar, 2020-03-03:
> My understanding is that the example failure is low FIRRTL that has run through optimizations and then can't be run back through the compiler?

Exactly. This is on the roadmap for "cleanup WIR" in #1350 and #1410, and duplicates a few other issues (#270, #866, #1171), so I am going to close this as "known issue, planned fix."

@aabdelsa, the issue is that you are dumping Low FIRRTL and then recompiling it. If you would like to dump your FIRRTL files and then compile them to Verilog in two separate steps, I would recommend emitting HighFirrtl in the first step. This will avoid this problem.

Implementing this solution is highly tied to your particular flow, so if you are using one of the open-source generator repos like Chipyard, this might be a good issue to open there.

closed 2020-03-03 by albert-magyar