chipsalliance/firrtl #1287: Feature Request: Allow printfs to go to different file descriptors

mwachs5, 2019-12-20:
### Checklist

- [*] Did you write out a description of the feature you want to see?
- [*] Did you look around for any related features?
- [*] Did you specify relevant external information?

### Feature Description

When emitting printf, don't hard-code the destination file descriptor to 80000002.

Instead use something like:
```
`ifndef PRINTFD
 `define PRINTF_FD 'h80000002.
`endif
```

Changing this bit of code:
    def printf(str: StringLit, args: Seq[Expression]): Seq[Any] = {
      val strx = str.verilogEscape +: args.flatMap(Seq(",", _))
-->      Seq("$fwrite(32'h80000002,", strx, ");")
    }

### Type of Feature

   - backend code generation

### Related Features

Yes this bit of code:
```
    def printf(str: StringLit, args: Seq[Expression]): Seq[Any] = {
      val strx = str.verilogEscape +: args.flatMap(Seq(",", _))
      Seq("$fwrite(32'h80000002,", strx, ");")
    }
```

### External Information

jackkoenig, 2019-12-20:
Seems reasonable, do you have any thoughts on finer granularity like allowing different printfs to go to different files? We don't need to conflate the issues, but if it's simple enough to kill two birds with one stone, might as well.

mwachs5, 2019-12-20:
hmm, not so much. It's really the user of the verilog who is going to have thoughts on where these end up.

seldridge, 2019-12-20:
@mwachs5: Do you want to change this after Verilog generation or is it something that should be Chisel-configurable?

I'm wondering if we need to add an extra define (which always seems kludgy) or if it should be an annotation that defines where things below it are sent. E.g., you annotate a module with stdout and some wire in that with stderr the prints get sent appropriately.

mwachs5, 2019-12-20:
I think it's unlikely that the Chisel author is going to know where things are supposed to be sent. This is for people actually simulating the verilog in their particular environment.

mwachs5, 2019-12-20:
I suppose a simple level of printf could also be useful, but that's getting fancy.

chick, 2020-01-12:
Noting related chisel issue: [Can chisel implement printf to a file #1290](https://github.com/freechipsproject/chisel3/issues/1290)

chick, 2020-01-16:
If this gets implemented it would be a good idea to update the following 
[stackoverflow: Can chisel implement printf to a file?
](https://stackoverflow.com/questions/59549805/can-chisel-implement-printf-to-a-file)