chipsalliance/firrtl #1290: Spec Clarification: info strings

wallento, 2019-12-30:
Hi,

just a quick question for clarification. As per the spec the info nodes contain string types. Despite not being explicitly specified strings are used in the spec with quotation marks, also when showing the info notation in listing 82.
I just ran the chisel tutorial from current releases and saw that info nodes are not in quotation marks. I am looking into adding a loFIRRTL frontend into Verilator and it would be nice to spare me some iterations ;)
Cheers,
Stefan

wallento, 2019-12-30:
I think it's preferable to have quotation marks, and to fix chisel to emit them

jackkoenig, 2019-12-30:
Yeah I think you're right. That is what the spec says; I think I just goofed when implementing this.

wallento, 2019-12-30:
Ah, right, I think I also tracked it down here: https://github.com/freechipsproject/chisel3/blame/e564fff139768e6a232d118ec7155ecf080f1bc4/chiselFrontend/src/main/scala/chisel3/internal/SourceInfo.scala#L46 which should probably be with the quotation marks, right?

jackkoenig, 2019-12-30:
Yeah that's the main offender. The other issue is to make sure we continue to parse the old style for backwards compatibility, but I really don't think it's too difficult.

Did you mean to close the issue?