chipsalliance/firrtl #1300: InferResets Emits Invalid FIRRTL

seldridge, 2020-01-08:
## Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?

The modifications to `InferResets` introduced in #1291 produce invalid FIRRTL that is only cleaned up after `ExpandWhens`. The implication of this is that any transform between `InferResets` and `ExpandWhens` is operating on invalid FIRRTL and has the potential to break. This impacts custom `HighForm` transforms. Additionally, this breaks #1123 as this adds a `CheckTypes` after `InferResets`.

@ucbjrl: This should block the 1.3 RC from going out today. :sob:

### What is the expected behavior?

You should be able to run `CheckTypes` after any transform and it should pass. If it doesn't pass, then it should pass with necessary resolution transforms, e.g., `InferTypes`. If it still doesn't pass, then the transform is broken (or there's a problem with resolution/checks!).

### Steps to Reproduce

Concretely, if you try to run `CheckTypes` after `InferResets` with #1291 applied, you wind up with an error. You can test this with the following diff:

```diff
diff --git a/src/main/scala/firrtl/LoweringCompilers.scala b/src/main/scala/firrtl/LoweringCompilers.scala
index 0d9b971b..9cbe6139 100644
--- a/src/main/scala/firrtl/LoweringCompilers.scala
+++ b/src/main/scala/firrtl/LoweringCompilers.scala
@@ -49,7 +49,9 @@ class ResolveAndCheck extends CoreTransform {
     new passes.TrimIntervals(),
     new passes.InferWidths,
     passes.CheckWidths,
-    new firrtl.transforms.InferResets)
+    new firrtl.transforms.InferResets,
+    passes.InferTypes,
+    passes.CheckTypes)
 }
 
 /** Expands aggregate connects, removes dynamic accesses, and when

```

### Your environment

- FIRRTL Versions failing: a4f2eda0 (current master), 3e096a73 (#1291 merge)
- FIRRTL Version passing: 70088cd2 (#1291 base)

### External Information

This was discussed between @seldridge and @jackkoenig on Slack.

### Discussion

This needs more discussion, but there's a question about whether or not the following should actually be legal (the #1291 example):

```
input a : UInt<1>
input b : AsyncReset
output z : Reset
z <= a
z <= b
```

Problematically, from the perspective of `CheckTypes` this is wrong as `z` can't have two types. This is similar to the following, which should continue to error:

```
input a: UInt<1>
input b: SInt<1>
output z: SInt<1>
z <= a
z <= b
```

It's not clear to me why this should be treated differently for resets as opposed to other ground types.

What I think needs to happen here is that `InferResets` needs to run on middle FIRRTL (where the IR is SSA and last-connect semantics are resolved) and not high FIRRTL. 

The current placement of `InferResets` is done, at least, to prevent deduplication of two modules with abstract reset that will later have resets inferred differently. One approach could involve running `EliminateTargetPaths` before a later `InferResets` to duplicate modules with abstract resets.

jackkoenig, 2020-01-08:
Another possible solution is to have `InferResets` do limited connection expansion and last connect semantics implementation. The basic idea is that `InferResets` can delete any offending connections, which is problematic if it's part of an aggregate connection, so it could selectively expand connections that are problematic but exclude problematic subconnections.

I think the `EliminateTargetPaths` approach is interesting, although I'm not a huge fan of how it destroys original Module names, perhaps we can solve that issue with annotations that keep the original names for instances.

seldridge, 2020-03-02:
Closing in favor of the solution in #1396.

closed 2020-03-02 by seldridge