chipsalliance/firrtl #1319: Stabilize Module Names

azidar, 2020-01-22:
### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

My idea is for a transform that attempts to rename modules to increase their relative stability across different Chisel generations.

It is important to note that this is a difficult problem to solve, because module names can conflict when generated, and any heuristic for disambiguating them makes them susceptible to changes in the seed for the heuristic.

For example, suppose we hash the contents of a module to use to disambiguate it from other modules of the same name. Now, any changes to that module's contents means its name changes; while effective at disambiguation, this coupling of the module's name to its contents means the module name is no longer an abstraction, as it depends on the internal contents. This may or may not be more desirable.

My suggestion is to use a sequence of heuristics to resolve collisions, where each heuristic reduces collisions but increases instability. If a set of collided modules are entirely disambiguated with one heuristic, the remaining heuristics are not used.

For now, my proposal for the sequence of heuristics, in the order they would be applied (if necessary), are as follows. This transform would apply post-deduplication.

1. Module name (chisel class name or user-specified name, tracked via annotation from Chisel)
  - Resolves simple modules instantiated once
2. Hash of IO structure (types, directions, but name-agnostic)
  - Resolves most parameterized modules
3. Hash of module contents (name-agnostic, child module name agnostic)
  - Resolves remaining parameterized modules, but robust to child module changes
4. Hash of module contents (including component and child module names)
  - Resolves all remaining differences between modules

### Type of Feature

   - backend code generation
   - new feature/API

### Related Features

Related to #1274, but for module names instead.

seldridge, 2020-01-22:
One other option may be to generate the name based on unambiguous hierarchical location derived from `Target`. E.g., a `ModuleTarget` if the only instance or an `InstanceTarget` with a path long enough to disambiguate. 

For background, I'm thinking of the [location-based naming schemes used by systemd for generating stable network interface names](https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/).

mwachs5, 2020-02-08:
I think the obvious complaint here is that adding a second module or changing parameters can drop you into a different scheme and change the module type.

Frankly, as a designer when *module* names change it's not that painful, if those changes can be localized somehow. Perhaps a more useful heuristic could be a hash of the hierarchical instance path to the module? Though of course that is not obvious how to dedup and would lead to potentially rippling effects.

seldridge, 2020-02-08:
Yeah, I was thinking of something derived from the hierarchy or a minimal path necessary to differentiate two modules. This was not thought out very far, but I was thinking of the stable device names in Linux like `wlp4s0` which are geographically derived. I was toying with thinking about something similar (however, the obvious difference is FIRRTL's "geography" is not fixed, as you state).

hcook, 2020-04-30:
It comes down to which you think is more likely to be resilient over time: the inner contents of a module or the outer hierarchical context of the module. I think the more "leaf-like" the module is, the more likely the former is to be stable than the latter, and the more "trunk-like" the module is, the more likely the latter is to be stable than the former. I am not sure if this insight is helpful but as I designer I think I could often tell you which kind a particular `Module` or `LazyModule` subclass is.

mwachs5, 2020-05-01:
Looking at this with fresh eyes I feel like @azidar's proposal is worth perusing to a prototype, I think the 'things oscillate when I add another one' is just sort of unavoidalbe and is at least unambiguous when you're doing a diff of before and after files. Or that's my hunch anyway

albertchen-sifive, 2020-05-01:
I wonder if these naming strategies could be exposed through some chisel API? like @hcook said, I think designers/Chisel would have better context and should be able to pick a preferred naming strategy, like a fuzzier version of `suggestName` that would emit something like `Use[Exact|IO|Contents|Path]NamingStrategyAnnotation`.

mwachs5, 2020-05-01:
I like that idea, would it have to be done in the module itself or could it be done by a user somewhere else? Something like "In this design, for all TLQueues, use Contents annotation"?

albertchen-sifive, 2020-05-01:
haven't really though it through, but I was thinking you could do both? like it would work kinda like `EliminateTargetPaths` where you could specify a naming strategy for any `IsModule` and any instances targeted by that `IsModule` would be duplicated using that naming strategy. So if you wanted all TLQueues to use the content hash then you could just emit that annotation from within the module constructor. or you could annotate individual instances at the instantiation site, and those instances would be duplicated together using the annotated naming strategy. and then maybe have some api that lets you choose a naming strategy for a sub-hierarchy. something like `HierarchyNamingScope { new Module1; new Module2 }` and all modules instantiated within that scope would be annotated with hierarchy naming strategy or something.

mwachs5, 2020-05-04:
Ok, So i think what you are suggesting is there is a default resolution algorithm like what @azidar defined originally, but then you can override it with the annotation that you are suggesting?

albertchen-sifive, 2020-05-04:
yeah