chipsalliance/firrtl #1338: infer and inline AND/OR bit-reduction

johningalls-sifive, 2020-01-27:
### Checklist
- [X] Did you write out a description of the feature you want to see?
- [X] Did you look around for any related features?
- [X] Did you specify relevant external information?

### Feature Description
It would be nice to reduce generated Verilog code size by reducing this emitted rtl:
```
assign b = a == 4'hf;
assign c = a != 4'h0;
assign d = a ? 4'hf : 4'h0;
```

to this emitted Verilog:
```
assign b = &a;
assign c = |a;
assign d = {4{a}};
```

### Type of Feature
- backend code generation

### Related Features
similar to https://github.com/freechipsproject/firrtl/pull/1270 and https://github.com/freechipsproject/firrtl/pull/1296

seldridge, 2020-01-27:
The first two should be a pretty straightforward substitution to `andr` and `orr` (which I forgot are in the FIRRTL spec). I agree with a transform to collapse that, but it may be more straightforward to first get Chisel3 to emit `andr`/`orr`/`xorr`.

E.g., I would expect the following to emit a FIRRTL `andr`, but it does not:
```scala
class Foo extends RawModule {
  val in = IO(Input(UInt(4.W)))
  val out = IO(Output(Bool()))
  out := in.andR
}
```

```
circuit Foo : 
  module Foo : 
    input in : UInt<4>
    output out : UInt<1>
    
    node _T = eq(in, UInt<4>("h0f"))
    out <= _T
```

What's the Chisel source generating the latter two? That might be a similar situation.

johningalls-sifive, 2020-01-28:
> What's the Chisel source generating the latter two? That might be a similar situation.

Verilog
```
assign d = a ? 4'hf : 4'h0;
assign e = a ? 4'h0 : 4'hf;
```

is emitted by Chisel
```
val a = Wire(Bool())
val d = Fill(4, a)
val e = Fill(4, !a)
```

seldridge, 2020-01-28:
@johningalls-sifive: For the `andR`/`orR`, take a look at #1315.

Thanks for the Chisel example showing the issue. [`util.Fill` is currently special casing the situation when the width of the `a` is one](https://github.com/freechipsproject/chisel3/blob/f1c4395bd608234fef5a60d8851036d1acb2382f/src/main/scala/chisel3/util/Bitwise.scala#L70). It may make sense to _not_ special case this and just emit concatenations. The concatenation emission could then be modified to special case the situation of emission when everything being concatenated together is the same.

closed 2021-04-05 by ingallsj