chipsalliance/firrtl #1340: Emit Combinational Logic as always @*

seldridge, 2020-01-27:
### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

The FIRRTL Verilog emitter currently has dramatically different behavior for register emission and combinational logic emission.

Registers will be structured in an `always` block which enables more semantically meaningful emission including:

1. `if`/`else if`/`else` (See #1204)
2. `switch` blocks (See freechipsproject/chisel3#1198)
3. (Eventual) grouping of conditions (See #1339). 

Conversely, combinational logic is emitted using `assign` with ternary muxes. 

This feature request proposes changing Verilog emission to emit in an `always @*` block to enable the "better" emission outlined in the enumerated list above.

### Type of Feature

- backend code generation

### Related Features

The existing method `regUpdate` contains most of the logic to do this for registers. That could be repurposed to work for combinational logic.

### External Information

No external info.

jackkoenig, 2020-02-04:
Something to keep in mind, `if-else` has drastically different behavior than ternary operators. If the condition to a ternary operator is `X`, the result depends on each path, but will be known if both bits are the same or `X` if they differ (eg. `f = X ? 00 : 01`, result is `0X`). On `if-else`, it just picks the else. For the same example `if (X) f=00; else f=01;`, you get `01` as the result.

Not a show stopper, we just would have to emit code to properly handle this (could be emitted as a macro to make the Verilog more readable).

mwachs5, 2020-02-08:
I am missing something in the explanation. What is the rationale for preferring always @(*) instead of assign statements?

jackkoenig, 2020-02-08:
I don't think it necessarily makes sense across the board, but there are places where it could help, like using `case` statements for state machines, or grouping signals that are assigned together (eg. flattened aggregates).

seldridge, 2020-03-02:
@jackkoenig: This may be breaking from a Verilog generation perspective, but I think it may be safer to *not* rely on this ternary emission. I don't think this is baked into Treadle or any other assumptions about what FIRRTL semantics mean. I admit this is thorny...

@mwachs5: The motivation is to improve Verilog readability. Long chains of ternaries have the same problem as long nested `if-else` chains: semantic information is lost. The verification task then gets harder. The case for Chisel is worse, etc. I posit that we should choose a readable solution over an unreadable solution in all situations so long as QoR is the same.

As a rough heuristic, I would propose using `always @*` emission for all situations that involve a mux with another mux in either it's true or false expressions. Currently, the FIRRTL Verilog emitter will just map combinational muxes to ternaries.

For more motivation, consider the following (which is the combinational equivalent of https://github.com/freechipsproject/chisel3/issues/1198#issue-500609980):

```scala
import chisel3._
import chisel3.util.{switch, is, SwitchContext}
import chisel3.stage.{ChiselGeneratorAnnotation, ChiselStage}

class Foo extends RawModule {
  val in = IO(Input(UInt(2.W)))
  val out = IO(Output(Bool()))

  out := {
    val tmp = WireDefault(false.B)
    when(in === 0.U)        { tmp := in(0) }
      .elsewhen(in === 1.U) { tmp := ~in(0) }
      .elsewhen(in === 2.U) { tmp := in(1) }
      .otherwise            { tmp := ~in(1)  }
    tmp
  }
}

(new ChiselStage)
  .execute(Array("-X", "verilog"), Seq(ChiselGeneratorAnnotation(() => new Foo)))
```

This currently produces:

```verilog
module Foo(
  input  [1:0] in,
  output       out
);
  wire  _T_1;
  wire  _T_3;
  wire  _T_6;
  wire  _GEN_0;
  wire  _GEN_1;
  assign _T_1 = in == 2'h0;
  assign _T_3 = in == 2'h1;
  assign _T_6 = in == 2'h2;
  assign _GEN_0 = _T_6 ? in[1] : ~in[1];
  assign _GEN_1 = _T_3 ? ~in[0] : _GEN_0;
  assign out = _T_1 ? in[0] : _GEN_1;
endmodule
```

In an ideal world, the designer wanted (case statements would be a bonus):

```verilog
module Foo(
  input  [1:0] in,
  output       out
);
  always @* begin
    if (in == 2'b00) begin
      out <= in[0];
    end else if (in == 2'b01) begin
      out <= ~in[0];
    end else if (in == 2'b10) begin
      out <= in[1];
    end else begin
      out <= ~in[1];
    end
  end
endmodule
```

albert-magyar, 2020-03-15:
> @jackkoenig: This may be breaking from a Verilog generation perspective, but I think it may be safer to _not_ rely on this ternary emission. I don't think this is baked into Treadle or any other assumptions about what FIRRTL semantics mean. I admit this is thorny...

Treadle doesn't rely on this because there are no Xs in Treadle. With two-state simulation and some underlying synchronous event model, a ternary always produces equivalent behavior to an if/else structure. We need to bake some assumptions about how to deal with four-state into the emitter not only out of practicality but because Verilog that is X-pessimal will cause broken simulations.

I think there are two related issues: we are constrained to produce code that is a refinement of the behavior specified by the FIRRTL input, but there is also a general issue that changing the emitted code while still satisfying the first condition can produce unforeseen consequences due to users' implicit assumptions. I imagine @jackkoenig  is talking about the second issue to some extent? One specific consequence is that removing ternaries would negate a large amount of the "bug-finding" power of randomization.

Of course, if we figure out how to remove all barriers to "all two-state, all the time" with concrete, randomized poison values for combinational logic, we won't need to worry about any of this :)