chipsalliance/firrtl #1356: Scala.js port

drom, 2020-02-05:
### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

Porting FIRRTL to Scala.js would enable a few new use cases:
1. Run FIRRTL on the webpage
2. Integrate FIRRTL with other JavaScript projects
3. Distribute FIRRTL over the NPM

### Type of Feature

- backend code generation

### Related Features

not found

### External Information

https://twitter.com/wavedrom/status/1225197219713703936

grebe, 2020-02-06:
I started this some time ago (see [this branch](https://github.com/grebe/firrtl/tree/scalajs)). I think the main work to be done is to build shims from java standard library to whatever backend you use. If targeting browser, you need to decide how you're dealing with files and other things. If targeting node, I think your choices are more straightforward, but it is still a considerable amount of work.

I think the JSON/YAML serialization stuff uses reflection extensively and is also therefore not a great candidate for ScalaJS. This mostly impacts how you'll deal with annotations. I think protobufs can be used with scalajs fairly easily, so this is probably the way to go.

drom, 2020-02-06:
Thank you @grebe for the update. Great to see work in progress, list of potential issues and idea about solutions. Do I understand that the major effort would be in:
* File IO replacement.
* JSON/YAML serialization without refection.

Any other issues on the horizon?

jackkoenig, 2020-02-06:
> JSON/YAML serialization without refection

How does one [de]serialize user-defined datatypes in an extensible way without reflection?

In any case, I think it's a moot point because _I think_ ScalaJS supports our needs for reflection via a macro annotation: https://www.scala-js.org/api/scalajs-library/latest/#scala.scalajs.reflect.annotation.EnableReflectiveInstantiation

itviewer, 2020-05-11:
I think we should better use c ++ to re-implement the firrtl tool, it can use the antlr4 Grammar directly.
so the firrtl tool no longer need a lot of dependencies(Scala,JVM ..), and it can be bound to whatever language.

sequencer, 2020-05-11:
> I think we should better use c ++ to re-implement the firrtl tool, it can use the antlr4 Grammar directly.

Maybe porting to Scala Native is more reasonable.