chipsalliance/firrtl #1371: Exception: "get valid points": shouldn't be here (related to AsyncReset)

mwachs5, 2020-02-07:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?

I get this error message, attempting to change some `Bool` into `Reset`. Specifically I changed the childReset in rocket-chip's LazyModule to a `Reset` instead of a `Bool` here:

https://github.com/chipsalliance/rocket-chip/blob/4f0cdea85c8a2b849fd582ccc8497892001d06b0/src/main/scala/diplomacy/LazyModule.scala#L194

```
class LazyRawModuleImp(val wrapper: LazyModule) extends RawModule with LazyModuleImpLike {
  // These wires are the default clock+reset for all LazyModule children
  // It is recommended to drive these even if you manually shove most of your children
  // Otherwise, anonymous children (Monitors for example) will not be clocked
  val childClock = Wire(Clock())
  val childReset = Wire(Bool()) <---- I changed this to Wire(Reset())
  childClock := Bool(false).asClock
  childReset := Bool(true)
  val (auto, dangles) = withClockAndReset(childClock, childReset) {
    instantiate()
  }
}
```

Then I got this error during FIRRTL elaboration (I am in an environment where I expect this to be getting driven by an AsyncReset):

```
======== Starting Transform firrtl.passes.ExpandConnects$ ========
Exception in thread "main" firrtl.FirrtlInternalException: Internal Error! get_valid_points: shouldn't be here - (UIntType(IntWidth(1)), AsyncResetType)
Please file an issue at https://github.com/ucb-bar/firrtl/issues
	at firrtl.Utils$.error(Utils.scala:423)
	at firrtl.Utils$.throwInternalError(Utils.scala:164)
	at firrtl.Utils$.get_valid_points(Utils.scala:472)
	at firrtl.Utils$.$anonfun$get_valid_points$2(Utils.scala:452)
	at scala.collection.IndexedSeqOptimized.foldLeft(IndexedSeqOptimized.scala:60)
	at scala.collection.IndexedSeqOptimized.foldLeft$(IndexedSeqOptimized.scala:68)
	at scala.collection.mutable.ArrayBuffer.foldLeft(ArrayBuffer.scala:49)
	at firrtl.Utils$.get_valid_points(Utils.scala:446)
	at firrtl.Utils$.$anonfun$get_valid_points$2(Utils.scala:452)
	at scala.collection.IndexedSeqOptimized.foldLeft(IndexedSeqOptimized.scala:60)
	at scala.collection.IndexedSeqOptimized.foldLeft$(IndexedSeqOptimized.scala:68)
	at scala.collection.mutable.ArrayBuffer.foldLeft(ArrayBuffer.scala:49)
	at firrtl.Utils$.get_valid_points(Utils.scala:446)
	at firrtl.Utils$.$anonfun$get_valid_points$2(Utils.scala:452)
	at scala.collection.IndexedSeqOptimized.foldLeft(IndexedSeqOptimized.scala:60)
	at scala.collection.IndexedSeqOptimized.foldLeft$(IndexedSeqOptimized.scala:68)
	at scala.collection.mutable.ArrayBuffer.foldLeft(ArrayBuffer.scala:49)
	at firrtl.Utils$.get_valid_points(Utils.scala:446)
	at firrtl.passes.ExpandConnects$.expand_s$1(Passes.scala:149)
	at firrtl.passes.ExpandConnects$.$anonfun$run$16(Passes.scala:163)
	at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:238)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
	at scala.collection.TraversableLike.map(TraversableLike.scala:238)
	at scala.collection.TraversableLike.map$(TraversableLike.scala:231)
	at scala.collection.AbstractTraversable.map(Traversable.scala:108)
	at firrtl.ir.Block.mapStmt(IR.scala:370)
	at firrtl.Mappers$StmtMagnet$$anon$1.map(Mappers.scala:16)
	at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:33)
	at firrtl.passes.ExpandConnects$.expand_s$1(Passes.scala:163)
	at firrtl.passes.ExpandConnects$.expand_connects$1(Passes.scala:168)
	at firrtl.passes.ExpandConnects$.$anonfun$run$19(Passes.scala:173)
	at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:238)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
[... 83 lines cut ...]
	at firrtl.Driver$.execute(Driver.scala:242)
	at firrtl.Driver$.main(Driver.scala:258)
	at firrtl.Driver.main(Driver.scala)
```

### What is the expected behavior?

The async reset should get connected to the childReset and propogated to the child modules as an async reset.

### Steps to Reproduce

See above.

### Your environment

- Chisel Verions: 
`* | 060b421 - (2 weeks ago) Fixed code example typo in comment (#1294) (#1313) - mergify[bot] (HEAD)`

- FIRRTL Version: 
`| * 52a1ee8 - (5 weeks ago) Infer resets last connect semantics (bp #1291) (#1292) - mergify[bot] (HEAD)`

- OS: 
- Verilator version: 

### External Information

seldridge, 2020-02-07:
Ugh... This appears to be dying in `ExpandConnects`. 

I think this is the infamous `InferResets` bug, #1300. The current implementation of `InferResets` is emitting invalid FIRRTL. `get_valid_points` is complaining because it sees a `UInt` driven by an `AsyncReset`. However, this is a connection which last-connect semantics would remove.

If you have the FIRRTL for this, would you be able to include it? I'm trying to know if this is #1300 or something else...

mwachs5, 2020-02-07:
"If you have the FIRRTL for this, would you be able to include it? I'm trying to know if this is #1300 or something else..."

Please excuse my ignorant question, but is this the FIrrtl at the very beginning or some intermediate stage?

mwachs5, 2020-02-07:
Seems likely it is the #1300 since the following line is that childReset := true.B and then later I know I am assigning it to an AsyncReset

seldridge, 2020-02-07:
**No need since you traced the signal.** Very beginning is fine (whatever Rocket Chip dumps out as the `*.fir` file).

Note: I may have a fix in place for this with #1301 which moves `InferResets` after `ExpandConnects` and `ExpandWhens` (meaning last connect semantics are resolved before resets are inferred so you never have two different concrete reset types driving each other). If you want to test this out, you could try the build with that FIRRTL fat jar. This is extremely preliminary, however.

mwachs5, 2020-02-07:
Note that commenting out the `childReset := false.B` had no effect so not sure that actually is the issue. Also looking for any assignments of `childReset` to `Async` I did not see:

```
grep childReset * | grep async
```
had no effect in the firrtl directory. I see lots of places `childReset` is getting assigned to UInt<1>. Any hint on how to diagnose from that error message which FIRRTL file is the culprit?

mwachs5, 2020-02-07:
(sorry can't include my entire firrtl file here)

seldridge, 2020-02-07:
Ignore #1301 for now. I still have Chisel tests failing on that.

> had no effect in the firrtl directory. I see lots of places childReset is getting assigned to UInt<1>. Any hint on how to diagnose from that error message which FIRRTL file is the culprit?

It think it's the reverse. It's saying that the LHS of a connection is of type `UInt` and the RHS is of type `AsyncReset`. That file is in chisel compatibility mode, so `Wire(Reset())` is really `WireDefault(Reset())` which I think is what is generating the initial connection.

To really narrow it down, you'd have to build a custom FIRRTL that dumps that line when the error happens and `get_valid_points` doesn't have enough context to actually do this.

> (sorry can't include my entire firrtl file here)

All good.

mwachs5, 2020-02-07:
"That file is in chisel compatibility mode, so Wire(Reset()) is really WireDefault(Reset()) which I think is what is generating the initial connection."

Ah, to be clear, this is the exact version of the file I am editing, which is sort-of-kindof-Chisel compatability mode:

https://github.com/chipsalliance/rocket-chip/blob/multiple-reset-schemes-rebased/src/main/scala/diplomacy/LazyModule.scala#L194

```
import Chisel.{defaultCompileOptions => _, _}
import freechips.rocketchip.util.CompileOptions.NotStrictInferReset
```

mwachs5, 2020-02-08:
Here's the part I am hypothesizing is causing the problem ( i think i cut this down to something that would compile but is just the relevant parts). I'm guessing its the line in the first file:
`    childReset <= _T_1.reset @[ClockDomain.scala 15:16]`
Again this is failing when the actual `childReset` being routed to that assignment through the complicated chain of clock assignments is actually an `AsyncReset`.

```
circuit FrontBus :
  extmodule ClockGroupAggregator_2 :
    output auto : { flip in : { member : { 0 : { clock : Clock, reset : Reset}}}, out : { member : { 0 : { clock : Clock, reset : Reset}}}}
  
    defname = ClockGroupAggregator_2
    

  extmodule ClockGroup_2 :
    output auto : { flip in : { member : { 0 : { clock : Clock, reset : Reset}}}, out : { clock : Clock, reset : Reset}}
  
    defname = ClockGroup_2
    

  extmodule FixedClockBroadcast_2 :
    output auto : { flip in : { clock : Clock, reset : Reset}, out : { clock : Clock, reset : Reset}}
  
    defname = FixedClockBroadcast_2

  module FrontBus :
    output auto : { flip front_bus_clock_groups_in : { member : { 0 : { clock : Clock, reset : Reset}}}
    output clock : Clock
    output reset : Reset
  
    auto is invalid
    clock is invalid
    reset is invalid
    wire childClock : Clock @[LazyModule.scala 194:24]
    childClock is invalid @[LazyModule.scala 194:24]
    wire childReset : Reset @[LazyModule.scala 195:24]
    childReset is invalid @[LazyModule.scala 195:24]
    node _T = asClock(UInt<1>("h0")) @[LazyModule.scala 196:29]
    childClock <= _T @[LazyModule.scala 196:14]
    childReset <= UInt<1>("h1") @[LazyModule.scala 197:14]
    inst front_bus_clock_groups of ClockGroupAggregator_2 @[BusWrapper.scala 31:48]
    front_bus_clock_groups.auto is invalid
    inst clockGroup of ClockGroup_2 @[BusWrapper.scala 32:38]
    clockGroup.auto is invalid
    inst fixedClockNode of FixedClockBroadcast_2 @[ClockGroup.scala 97:107]
    fixedClockNode.auto is invalid

    wire _T_1 : { clock : Clock, reset : Reset} @[Nodes.scala 370:76]
    _T_1 is invalid @[Nodes.scala 370:76]
    clockGroup.auto.in <- front_bus_clock_groups.auto.out @[LazyModule.scala 168:57]
    fixedClockNode.auto.in <- clockGroup.auto.out @[LazyModule.scala 168:57]
    front_bus_clock_groups.auto.in <- auto.front_bus_clock_groups_in @[LazyModule.scala 174:31]
    childClock <= _T_1.clock @[ClockDomain.scala 14:16]
    childReset <= _T_1.reset @[ClockDomain.scala 15:16]
    clock <= _T_1.clock @[ClockDomain.scala 21:11]
    reset <= _T_1.reset @[ClockDomain.scala 22:11]
```

```
circuit ClockGroupAggregator_2 :
  module ClockGroupAggregator_2 :
    output auto : { flip in : { member : { 0 : { clock : Clock, reset : Reset}}}, out : { member : { 0 : { clock : Clock, reset : Reset}}}}
  
    auto is invalid
    wire childClock : Clock @[LazyModule.scala 194:24]
    childClock is invalid @[LazyModule.scala 194:24]
    wire childReset : Reset @[LazyModule.scala 195:24]
    childReset is invalid @[LazyModule.scala 195:24]
    node _T = asClock(UInt<1>("h0")) @[LazyModule.scala 196:29]
    childClock <= _T @[LazyModule.scala 196:14]
    childReset <= UInt<1>("h1") @[LazyModule.scala 197:14]
    wire in_0 : { member : { 0 : { clock : Clock, reset : Reset}}} @[Nodes.scala 370:76]
    in_0 is invalid @[Nodes.scala 370:76]
    wire out_0 : { member : { 0 : { clock : Clock, reset : Reset}}} @[Nodes.scala 369:76]
    out_0 is invalid @[Nodes.scala 369:76]
    auto.out <- out_0 @[LazyModule.scala 174:49]
    in_0 <- auto.in @[LazyModule.scala 174:31]
    out_0.member.0.reset <= in_0.member.0.reset @[ClockGroup.scala 50:60]
    out_0.member.0.clock <= in_0.member.0.clock @[ClockGroup.scala 50:60]
```

```
circuit ClockGroup_2 :
  module ClockGroup_2 :
    output auto : { flip in : { member : { 0 : { clock : Clock, reset : Reset}}}, out : { clock : Clock, reset : Reset}}
  
    auto is invalid
    wire childClock : Clock @[LazyModule.scala 194:24]
    childClock is invalid @[LazyModule.scala 194:24]
    wire childReset : Reset @[LazyModule.scala 195:24]
    childReset is invalid @[LazyModule.scala 195:24]
    node _T = asClock(UInt<1>("h0")) @[LazyModule.scala 196:29]
    childClock <= _T @[LazyModule.scala 196:14]
    childReset <= UInt<1>("h1") @[LazyModule.scala 197:14]
    wire in : { member : { 0 : { clock : Clock, reset : Reset}}} @[Nodes.scala 370:76]
    in is invalid @[Nodes.scala 370:76]
    wire out_0 : { clock : Clock, reset : Reset} @[Nodes.scala 369:76]
    out_0 is invalid @[Nodes.scala 369:76]
    auto.out <- out_0 @[LazyModule.scala 174:49]
    in <- auto.in @[LazyModule.scala 174:31]
    out_0.reset <= in.member.0.reset @[ClockGroup.scala 25:52]
    out_0.clock <= in.member.0.clock @[ClockGroup.scala 25:52]
```

```
circuit FixedClockBroadcast_2 :
  module FixedClockBroadcast_2 :
    output auto : { flip in : { clock : Clock, reset : Reset}, out : { clock : Clock, reset : Reset}}
  
    auto is invalid
    wire childClock : Clock @[LazyModule.scala 194:24]
    childClock is invalid @[LazyModule.scala 194:24]
    wire childReset : Reset @[LazyModule.scala 195:24]
    childReset is invalid @[LazyModule.scala 195:24]
    node _T = asClock(UInt<1>("h0")) @[LazyModule.scala 196:29]
    childClock <= _T @[LazyModule.scala 196:14]
    childReset <= UInt<1>("h1") @[LazyModule.scala 197:14]
    wire in : { clock : Clock, reset : Reset} @[Nodes.scala 370:76]
    in is invalid @[Nodes.scala 370:76]
    wire out_0 : { clock : Clock, reset : Reset} @[Nodes.scala 369:76]
    out_0 is invalid @[Nodes.scala 369:76]
    auto.out <- out_0 @[LazyModule.scala 174:49]
    in <- auto.in @[LazyModule.scala 174:31]
    out_0.reset <= in.reset @[ClockGroup.scala 91:21]
    out_0.clock <= in.clock @[ClockGroup.scala 91:21]
```

mwachs5, 2020-02-08:
Hmm, interesting. I changed the Chisel based on @seldridge 's comment about compatability mode to:

```
// If we use a Chisel.Wire, then this gets WireDefault assigned to a Bool, which
  // we can't override later to an AsyncReset. Explicitly define this as a chisel3.Wire.
  // Also, do NOT drive it with a default value here. This will result in a Chisel elab
  // error if someone does not actually drive it, vs waiting to see a failing sim.
  val childReset = chisel3.Wire(Reset())
  childClock := Bool(false).asClock
  //childReset := Bool(true)
  val (auto, dangles) = withClockAndReset(childClock, childReset) {
    instantiate()
  }
}
```

And now the FIRRTL does NOT initialize childReset to anything, but I still get the same error. So I may be chasing the wrong thing here.

mwachs5, 2020-02-10:
Given that I have a `.pb` file of my entire design and a directory of the split-up `.fir` files, is there a way to just run firrtl on a subset of the design? So that I might be able to isolate what part of the hierarchy is causing the problem? I am looking for an exact firrtl command that I might be able to run to help narrow this down...

seldridge, 2020-02-10:
I think you can just use the `-i Foo.fir` command to compile some FIRRTL IR into Verilog. There's decent help text if you give this `--help`, too.

There's then a lot of options for how to do this:

1. If you have Rocket Chip, you have have a fat jar in `firrtl/utils/bin/firrtl.jar`. There's a helper script to use this, so you can do:

```bash
cd $MyRocketClone
./firrtl/utils/bin/firrtl -i Foo.fir
# Note: this requires Rocket Makefile magic to have built this fat jar...
```

2. You can clone FIRRTL and run it with `sbt`:

```bash
cd $MyFirrtlClone
sbt "runMain firrtl.stage.FirrtlMain -i Foo.fir"
```

3. You can build the fat jar manually and use that:

```bash
cd $MyFirrtlClone
sbt assembly
./utils/bin/firrtl -i Foo.fir
```

4. If you only need a tagged version, you can always use coursier:

```bash
coursier launch edu.berkeley.cs:firrtl_2.12:1.2.2 -M firrtl.stage.FirrtlMain -- -i Foo.fir
```

If what you're doing is very Rocket Chip-specific, the exact invocation that the open-source version is using for the emulation is [on this line in emulator/Makefrag-verilator](https://github.com/chipsalliance/rocket-chip/blob/4ca9e4debe580ae500b364bbd9544a0c1fcbb824/emulator/Makefrag-verilator#L17).

Apologies if you already know this...

mwachs5, 2020-02-21:
This is resolved on the 1.2.x branch

closed 2020-02-21 by mwachs5