chipsalliance/firrtl #1379: Expecting WDefInstance, found a DefInstance

ShuyunJia, 2020-02-11:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [ ] Did you specify relevant external information?

### What is the current behavior?
I'm trying to add some transform in rocket-chip generator. Since the circuit is elaborated in `chisel3.internal.Circuit`, I use the `chisel3.Driver.toFirrtl` to convert to `firrtl.ir.Circuit`. Here's my code sample:
```
def applyTransform : Unit = {
  val transform = new TopWiringTransform
  transform.execute(firrtl.CircuitState(Driver.toFirrtl(circuit), firrtl.MidForm))
}
```
I got the following erro msg:
```
[error] (run-main-0) firrtl.FirrtlInternalException: Internal Error! Expecting WDefInstance, found a DefInstance!
[error] Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error] firrtl.FirrtlInternalException: Internal Error! Expecting WDefInstance, found a DefInstance!
[error] Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error] 	at firrtl.Utils$.error(Utils.scala:423)
[error] 	at firrtl.Utils$.throwInternalError(Utils.scala:164)
[error] 	at firrtl.analyses.InstanceGraph$.collectInstances(InstanceGraph.scala:153)
[error] 	at firrtl.analyses.InstanceGraph$.$anonfun$collectInstances$1(InstanceGraph.scala:155)
[error] 	at firrtl.analyses.InstanceGraph$.$anonfun$collectInstances$1$adapted(InstanceGraph.scala:155)
[error] 	at scala.collection.Iterator.foreach(Iterator.scala:929)
[error] 	at scala.collection.Iterator.foreach$(Iterator.scala:929)
[error] 	at scala.collection.AbstractIterator.foreach(Iterator.scala:1417)
[error] 	at scala.collection.IterableLike.foreach(IterableLike.scala:71)
[error] 	at scala.collection.IterableLike.foreach$(IterableLike.scala:70)
[error] 	at scala.collection.AbstractIterable.foreach(Iterable.scala:54)
[error] 	at firrtl.ir.Block.foreachStmt(IR.scala:375)
[error] 	at firrtl.traversals.Foreachers$StmtForMagnet$$anon$1.foreach(Foreachers.scala:18)
[error] 	at firrtl.traversals.Foreachers$StmtForeach$.foreach$extension(Foreachers.scala:35)
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 29 s, completed Feb 11, 2020 12:55:41 PM
```
### What is the expected behavior?
able to run the transform

### Steps to Reproduce

1. add `def applyTransform` in 
https://github.com/chipsalliance/rocket-chip/blob/176c5e1d444c57a99ffe922127955599edbe8027/src/main/scala/util/GeneratorUtils.scala#L109
2. add `applyTransform` in
https://github.com/chipsalliance/rocket-chip/blob/176c5e1d444c57a99ffe922127955599edbe8027/src/main/scala/system/Generator.scala#L105
3. make

### Your environment

- Chisel Verions: <-- 3.2.0 -->
- OS: <-- Linux version 4.4.0-142-generic (buildd@lgw01-amd64-033) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.10) ) #168-Ubuntu SMP Wed Jan 16 21:00:45 UTC 2019
 -->
- Verilator version: <-- 4.008 -->

### External Information

I'm not sure yet

albert-magyar, 2020-02-11:
Like many transforms, the `TopWiringTransform` uses "working IR," which augments some nodes of the in-memory IR representation with several analysis fields for convenience. A `WDefInstance` is just an `Instance` as represented in "working IR," where it carries around the type it would have when referenced.

The fact that these forms are distinct and that a pass cannot interchangably use them in the cases where it does not care to look at those analyses is fixed in #1350, which is under review for FIRRTL 1.4.0.

For now, the solution is to convert the `CircuitState` to working IR by running `firrtl.IRToWorkingIR`, which is a core transform.

albert-magyar, 2020-02-11:
Another potential issue is that you're labeling starting circuit as `MidForm` when it very well may be `HighForm`. A more robust way to add a custom transform would be to use the `custom-transforms` option in your FIRRTL flow.

closed 2020-02-11 by albert-magyar