chipsalliance/firrtl #1393: Div and Rem act as sources of X

albert-magyar, 2020-02-13:
I ran into this problem when trying to run LEC to test examples for #1365.

From IEEE 1364-2005 (same for older standards):
>For the division or modulus operators, if the second operand is a zero, then the entire result value shall be x.

I know these don't pop up too much in practice, but it might be reasonable to control via randoms.
```
z <= a / b;                        // source of X
z <= a % b;                        // source of X
z <= (b == 0) ? _RAND_0 : a % b;   // no X
z <= (b == 0) ? _RAND_0 : a / b;   // no X
```

albert-magyar, 2020-02-13:
This also means that `Div` and `Rem` _always_ produce `1'bX` for zero-width arguments.

jwright6323, 2020-02-13:
Is this meant to be the in a synthesizeable code block produced by FIRRTL? This would generate additional hardware. Having them return 0 is probably less hardware, but still more than nothing (which leaves the synthesis tool to do as it pleases).

aswaterman, 2020-02-13:
Can use an ifdef-guard in the same manner as out-of-bounds array indices,
so that the new Verilog only affects simulation, letting the synthesis tool
do as it pleases.

albert-magyar, 2020-02-13:
> Is this meant to be the in a synthesizeable code block produced by FIRRTL? This would generate additional hardware. Having them return 0 is probably less hardware, but still more than nothing (which leaves the synthesis tool to do as it pleases).

Sorry I didn't specify; this would be simulation only -- the `_RAND_` signals are generally only used in simulation. This would fit in our existing set of conditional compilation modes that cover things like register initialization.

ekiwi, 2020-02-24:
Can we add wording about division by zero leading to an arbitrary result to the firrtl spec?

Also: strictly speaking, the result of division by zero should be re-randomized every cycle to account for the allowable worst-case, right?

albert-magyar, 2020-02-24:
> Can we add wording about division by zero leading to an arbitrary result to the firrtl spec?

Sure, I opened #1409 for that.

> Also: strictly speaking, the result of division by zero should be re-randomized every cycle to account for the allowable worst-case, right?

That is the way any randomization based on an undefined operator result would be implemented, though it represents a practical test rather than a theoretical worst case (combinational logic has no notion of cycles).