chipsalliance/firrtl #1406: Way to specify top Circuit's Reset type

mwachs5, 2020-02-21:
### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

Right now, if I understand correctly, there is no way to specify that a top-level Circuit which has reset type of `Reset` should actually have an async reset type, the assumption is that it wants Bool. Short of modifying the firrtl, is there a way to do this? Is there a simple transform that could be run to do this, given some annotation?

### Type of Feature

 - new feature/API 

### Related Features

Async Reset
### External Information

johnsbrew, 2020-02-21:
I might misunderstand your question but there is no issue with specifying `AsyncReset` as top-level reset.

See https://github.com/freechipsproject/firrtl/blob/master/src/test/scala/firrtlTests/AsyncResetSpec.scala#L22

If you mean that to change one in another, this operation is not meant to be done by firrtl but rather during generation. Hence with `withReset` statement within your chisel top level.

mwachs5, 2020-02-21:
withReset in chisel top level doesn't really work for a Module in general. I agree that there should be an API for a module to decide whether it is sync or async reset (without having to wrap everything in withReset, which is quite inflexible), but that API doesn't currently exist in Chisel. It's only a module's instantiator that gets decide whether its `reset` is Bool or Async. 

However, I realize now that the emitted chisel in my case (which is coming from rocket-chip) is indeed setting the top-level reset to UInt<1> in the emitted firrtl, so this request may be rather moot, and better redirected to chisel repo.

mwachs5, 2020-02-21:
but to clarify my original question, say that i hand-wrote a firrtl circuit like this:

```
circuit MyTopLevelModule:
  module MyTopLevelModule:
    input clock : Clock
    input reset : Reset

     ... some register logic here
```

First of all, is that legal firrtl? And if so, my understanding is after InferResets it will assume that reset needs to be Bool. My question is it possible to change that assumption with some sort of annotation/change to InferResets, or must I figure out how to make it emit from chisel as the proper type (with the limitations on chisel explained above).

jackkoenig, 2020-02-21:
Currently FIRRTL disallows abstract reset (aka `Reset`) for top-level inputs or blackbox outputs. There clearly needs to be some way to express a default and a per-port override because otherwise you can't split up FIRRTL circuits into multiple pieces (my favorite technique for debugging).