chipsalliance/firrtl #1410: RFC: Should Addw/Subw/Dshlw become "first-class" IR nodes?

albert-magyar, 2020-02-24:
Now that #1350 is going to remove the redundancy of WIR nodes that correspond with "augmented" IR nodes in 1.4.0, we should figure out to do with the other WIR nodes.

**Some (`Addw`/`Subw`/`Dshlw`) are actually distinct and useful flavors of IR nodes that have slightly different width semantics.** I think these are important and we might want to consider including them in the IR.

Other WIR nodes are just random pass-internal pseudo-nodes that people stashed in `WIR.scala`. I think they should be split out to different places in the code to represent their "lifetimes" in the compiler.

`WDefInstanceConnector` => only used in `VerilogPrep` and `VerilogEmitter`
`WVoid`, `WInvalid` => become private to `ExpandWhens` and perhaps a couple core transforms
`EmptyExpression` => becomes private to `passes.memlib`
`CDefMPort`, `CDefMemory` => handled in #1210

albert-magyar, 2020-03-03:
Argument in favor of inclusion: currently, non-growing ops are used as a bridge to Verilog. However, they also make it possible to match non-growing Chisel operations with a single op.

Argument against inclusion: more code complicates the compiler.

jackkoenig, 2020-03-03:
Hedge against compiler complication is the notion of "canonical form" where you choose a subset of nodes and allow transform writers to write against this form. I think it is perfectly sensible to create a "canonical form" as a subset of primops in LowForm to make it easier to write various transforms.

albert-magyar, 2020-03-03:
I agree. I think the odd thing is that it could currently be argued that `Addw` et al. are the "canonical form" despite being absent in the spec :)

jackkoenig, 2020-03-03:
I would say they're not canonical form because they can't express expanding addition whereas `Add` + `Tail` can express `Addw`, it's not unusually to go from canonical form to a richer set in order to make emission prettier

albert-magyar, 2020-03-03:
> I would say they're not canonical form because they can't express expanding addition whereas `Add` + `Tail` can express `Addw`

I mean they CAN ... but your point is a great one nonetheless.

grebe, 2020-03-03:
It seems to me `addw ≡ tail(add(a, b), 1)`, and the inlined form could be treated as `Addw` in the parts of the compiler where you care, but otherwise not need any other sort of support. Are inlined operators like this unfriendly to firrtl as currently implemented?

seldridge, 2020-03-03:
I'm against these for the sole reason that they exist in the FIRRTL compiler to represent the semantics of a single backend (Verilog). As new backends and front ends proliferate, it seems more natural to have a canonical IR as opposed to allowing the set of IR nodes to increase, though that would be a tempting option. I admit that *w operators straddle this boundary as they do seem useful and exist in Chisel (though I wouldn't let the Chisel API guide the IR nodes selection).

A better approach, albeit enormously more complicated, would be to explore a notion of "dialects" in FIRRTL that would allow new operators to be added. The necessary support for this would likely be another enormous infrastructure refactor.

albert-magyar, 2020-03-03:
> It seems to me `addw ≡ tail(add(a, b), 1)`, and the inlined form could be treated as `Addw` in the parts of the compiler where you care, but otherwise not need any other sort of support. Are inlined operators like this unfriendly to firrtl as currently implemented?

There are places in the compiler where these kinds of multi-level expression trees end up getting divided, but they are generally supported, and there isn't any pass between `PadWidths` and `VerilogEmitter` that would break them up.

The approach you're describing is what I'd like to do if we decide against including the `Xw` operators in the IR. It's already the strategy `CombineCats` uses to improve Verilog emission.

albert-magyar, 2020-03-03:
> It seems to me `addw ≡ tail(add(a, b), 1)`, and the inlined form could be treated as `Addw` in the parts of the compiler where you care, but otherwise not need any other sort of support. Are inlined operators like this unfriendly to firrtl as currently implemented?

One nice new thing is that #1241 made it strictly true that `addw ≡ tail(add(a, b), 1)`. This was not the case before that PR if `a` and `b` were both zero-width. I think this change is an extra point in favor of dropping `Addw` et al.

albert-magyar, 2020-03-03:
So it currently seems like nobody actually wants these as first-class IR nodes. I am certainly not a fan of adding more stuff if it's not necessary.

In a similar vein, I am generally in favor of the idea of "custom IR nodes." However, it seems like the general feeling is that **pattern-matched op-tail subtrees** could provide identical benefits in the narrow context where `Dshlw` et al. are used, without the potential for breaking downstream code. This pattern-matched subtree approach is also used by `CombineCats`.

To me, this makes it seem like the current use of `Dshlw` isn't a great motivating example for custom ops. I think we should wait to have a great motivating use case before adding new APIs that potentially affect all transforms.

My current vote for fixing the many wrap-op issues is eliminating them for 1.4.0. I already have some code for replacing them with pattern-matched subtrees, and I can roll that into #1350.

seldridge, 2020-03-03:
I think custom IR nodes makes a lot of sense. :+1: `addw` and friends may, as you suggest, be too simple of a motivating case. I *think* this is because their rewrite rules are pretty simple. I.e., an `Expression => Expression`.

More complex examples in terms of types might, in order of complexity, include:

1. `FixedPoint` (How could we rewrite `FixedPointToSInt` and `PadWidths` where the special sauce for the custom nodes is encoded not in the transforms, but in the actual custom node?)
1. Floating Point (Transforming `1.0 + 2.0` would, at minimum, be `Expression => Statement` and likely require emission of an FPU instance to handle this. However, it seems very useful to be able to just pass around floating point types and lower them later.)
1. Tensors

More complex ops could include:

1. Vector operations
1. Cross module references

I expect the complexity here is that we need to think about how to generalize some transforms to work on custom types where the custom types define how transforms use them. "Common" transforms should likely work across all types and custom types, i.e., Common Subexpression Elimination, Dead Code Elimination, (to-be-written Global Value Numbering), Inlining, Flattening, etc. At minimum, anything custom would need to define how it is lowered to canonical IR. I think that then existing transforms would have to know to ignore custom IR nodes and conservatively excluded them from any optimizations.

tl;dr: enormous infrastructure rabbit hole...

albert-magyar, 2020-03-03:
> I think custom IR nodes makes a lot of sense. 👍 `addw` and friends may be, as you suggest, too simple of a motivating case. I _think_ this is because their rewrite rules are pretty simple. I.e., an `Expression => Expression`.

Exactly. My concern with `addw` is not the work in adding the infrastructure, but that the infrastructure invariably takes some flavor from the set of pre-existing motivating use cases, which would possibly lead to some contrived decisions if those were only the wrap ops.

grebe, 2020-03-03:
Not to pull the conversation in a random direction, but I these custom IR nodes may have some similar flavor to a pattern I've been running in to:

1) Start off with `a := f(b)` being implemented by making an `ExtModule` (or empty `DefModule` with `DontTouch`, whatever, it's just a placeholder to block optimization passes until the implementation gets filled in), with `a` and `b` as outputs and inputs to the module.
2) Run the FIRRTL compiler until things have been lowered to a point that is convenient to run whatever analysis is needed
3) Replace the `ExtModule` with some implementation, which may also get inlined, optimized, etc.

I think this pattern could apply to custom IR nodes. It's a little silly for something like `Addw` because we can immediately fill in the implementation, but for more sophisticated things I think it is helpful. `FixedPoint` is actually a great example. You could have a `FixedPoint` be an `SInt` with an annotation for it's binary point, emit a bunch of placeholder modules for the various operations, and then a transform comes in and implements the logic for doing those operations.
The tricky bit would be dealing with inference, but I think having an API for adding constraints could be generically useful (e.g. marking an `ExtMod` as having an output width that is a function of both input width and binary point).