chipsalliance/firrtl #1417: Inlining Should Add Parentheses

seldridge, 2020-03-03:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [ ] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?

For all the operator inlining transforms, no parentheses are inserted. This may cause problems with different backends (e.g., Verilog/VHDL) where the semantics of operators are different.

### What is the expected behavior?

Parentheses should be included. There's no notion of parentheses in FIRRTL, but we could use `ir.Block` to represent this.

### Steps to Reproduce

(On my phone, so no example.)

Write down FIRRTL that includes nodes with nots or bit extractions (something that will be inlined).

### Your environment

Master

### External Information

None.

albert-magyar, 2020-03-10:
I think the inlining of real operators is being killed for the time being, as it needs a total rework to avoid breaking Verilog context-dependent widths.

I would say that anything where parens would be helpful is certainly going to cause other problems (we're certainly not going to inline binary ops at all without a larger Emitter rework), so I think we can close this for the time being?

seldridge, 2020-03-10:
Good to shelve. We can re-open once inlining is more broadly on the table for discussion.

closed 2020-03-10 by seldridge