chipsalliance/firrtl #1466: Turn off coverage gathering for simulation-only initial blocks

mwachs5, 2020-03-22:
### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information? -- *Not aware of any*

### Feature Description

During code emission, it would be nice to turn off (optionally) coverage for the random initialization code initial blocks. These constitute a large portion of lines of code and are difficult to cover since they are randomly initialized.
For certain tool flows it can be turned off by wrapping the  `initial block` with `// VCS coverage off` and `// VCS coverage on`

Similar to the addition of `ifndef SYNTHESIS`:
https://github.com/freechipsproject/firrtl/blob/5c0c0018d812d57270035a9d3bd82e2289acf4ec/src/main/scala/firrtl/Emitter.scala#L717

If done as requested, it would look like the following:
```
`ifndef SYNTHESIS
// VCS coverage off
initial begin
...

end // initial
// VCS coverage on
`endif // SYNTHESIS
```

### Type of Feature

### Related Features

Similar to the code for `ifndef SYNTHESIS added here: 
https://github.com/freechipsproject/firrtl/blob/5c0c0018d812d57270035a9d3bd82e2289acf4ec/src/main/scala/firrtl/Emitter.scala#L717

### External Information

mwachs5, 2020-04-20:
perhaps this should be expanded for generic `BEFORE_INITIAL and `AFTER_INITIAL macros which can be populated with a variety of macros by the final verilog user

debs-sifive, 2020-05-05:
@mwachs5 can this be closed, since #1550 has been merged?

mwachs5, 2020-05-05:
Closed by #1550 🎉

closed 2020-05-05 by mwachs5