chipsalliance/firrtl #1492: unidoc generation fails with Scala 2.11.12 on JDK 11

ucbjrl, 2020-04-01:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?
```
$ sbt ++2.11.12 unidoc
```
on Java 11 fails with:
```
[info] Setting Scala version to 2.11.12 on 1 projects.
[info] Reapplying settings...
[info] Set current project to firrtl (in build file:.../firrtl/)
[info] Main Scala API documentation to .../firrtl/target/scala-2.11/unidoc...
model contains 902 documentable templates
[error] .../firrtl/src/main/scala/firrtl/options/Exceptions.scala:11:1: Could not find any member to link for "java.lang.Throwable".
[error] /** Indicate an error related to a bad [[firrtl.annotations.Annotation Annotation]] or it's command line option
[error] ^
[error] one error found
[error] (Scalaunidoc / doc) Scaladoc generation failed
[error] Total time: 61 s (01:01), completed Apr 1, 2020, 3:14:16 PM
```

### What is the expected behavior?

### Steps to Reproduce
```
$ sbt ++2.11.12 unidoc
```

### Your environment

- OS: Darwin ucbjrlMBP.local 18.7.0 Darwin Kernel Version 18.7.0: Thu Jan 23 06:52:12 PST 2020; root:xnu-4903.278.25~1/RELEASE_X86_64 x86_64
- JDK:openjdk version "11" 2018-09-25
OpenJDK Runtime Environment 18.9 (build 11+28)
OpenJDK 64-Bit Server VM 18.9 (build 11+28, mixed mode)

### External Information
NOTE: This issue isn't specific to FIRRTL.

https://github.com/scala/bug/issues/10675

jackkoenig, 2020-04-01:
We don't publish the 2.11 Scaladoc anymore so maybe this is okay to ignore until we drop 2.11 support?

ucbjrl, 2020-04-01:
The problem is it also breaks 2.11 publishing on JDK11. I think it's a side-effect of #1471. We can work around it, but it does increase the pressure to get 2.13 support working and drop support for 2.11.

jackkoenig, 2020-04-01:
Oh yeah that is a problem