chipsalliance/firrtl #1500: Internal Error - DeadCodeElimination

pentin-as, 2020-04-07:
[error] (run-main-0) firrtl.FirrtlInternalException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error] firrtl.FirrtlInternalException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error]         at firrtl.Utils$.error(Utils.scala:445)
[error]         at firrtl.Utils$.throwInternalError(Utils.scala:164)
[error]         at firrtl.stage.FirrtlStage.run(FirrtlStage.scala:35)
[error]         at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error]         at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:32)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Translator.transform(Phase.scala:140)
[error]         at firrtl.options.Translator.transform$(Phase.scala:140)
[error]         at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:46)
[error]         at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error]         at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error]         at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] Caused by: java.util.NoSuchElementException: None.get
[error]         at scala.None$.get(Option.scala:529)
[error]         at scala.None$.get(Option.scala:527)
[error]         at firrtl.transforms.DeadCodeElimination.createDependencyGraph(DeadCodeElimination.scala:174)
[error]         at firrtl.transforms.DeadCodeElimination.run(DeadCodeElimination.scala:297)
[error]         at firrtl.transforms.DeadCodeElimination.execute(DeadCodeElimination.scala:351)
[error] apeout /at firrtl.Transform.$anonfun$runTransform$2(Compiler.scala:218)
[error] 	at firrtl.Utils$.time(Utils.scala:169)
[error] 	at firrtl.Transform.runTransform(Compiler.scala:218)
[error] 	at firrtl.SeqTransformBased.$anonfun$runTransforms$1(Compiler.scala:283)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at firrtl.SeqTransformBased.runTransforms(Compiler.scala:283)
[error] 	at firrtl.SeqTransformBased.runTransforms$(Compiler.scala:282)
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 1425 s (23:45), completed 07.04.2020 10:56:15

firrtl: eb637777e3c4d77435cfd13358c521ed1b766ba8
chisel: 7a343dce95a370f6cb7b9cf80e0694ac82dc94f8

I think I have a problem in my code. But since there was a FIRRTL error, I create an issue.

I call FIRRTL as follows:
running barstools.tapeout.transforms.GenerateTopAndHarness -o .../chipyard.TitanHarness.TitanConfig.top.v -tho .../chipyard.TitanHarness.TitanConfig.harness.v -i .../chipyard.TitanHarness.TitanConfig/chipyard.TitanHarness.TitanConfig.fir --syn-top TitanSystem --harness-top TestHarness -faf .../chipyard.TitanHarness.TitanConfig.anno.json -tsaof .../chipyard.TitanHarness.TitanConfig.top.anno.json -tdf .../firrtl_black_box_resource_files.top.f -tsf .../chipyard.TitanHarness.TitanConfig.top.fir -thaof .../chipyard.TitanHarness.TitanConfig/chipyard.TitanHarness.TitanConfig.harness.anno.json -hdf .../chipyard.TitanHarness.TitanConfig/firrtl_black_box_resource_files.harness.f -thf .../chipyard.TitanHarness.TitanConfig.harness.fir --infer-rw --repl-seq-mem -c:TitanHarness:-o:.../chipyard.TitanHarness.TitanConfig.top.mems.conf -thconf .../chipyard.TitanHarness.TitanConfig.harness.mems.conf -td .../chipyard.TitanHarness.TitanConfig

Here is my .fir file:
[chipyard.TitanHarness.TitanConfig.fir.zip](https://github.com/freechipsproject/firrtl/files/4442765/chipyard.TitanHarness.TitanConfig.fir.zip)

pentin-as, 2020-04-07:
Specified an invalid value with the --harness-top key.

closed 2020-04-07 by pentin-as

jackkoenig, 2020-04-08:
Glad you were able to resolve it. Not ideal that an incorrect CLI argument lead to an internal error though...